cocotb/cocotb #1296: Swig is missing from CI

eric-wieser, 2020-01-15:
From [this CI run](https://travis-ci.org/cocotb/cocotb/jobs/635275148#L4128-L4139):
```
make[2]: Entering directory '/home/travis/build/cocotb/cocotb/examples/endian_swapper/tests'
/usr/bin/make results.xml
make[3]: Entering directory '/home/travis/build/cocotb/cocotb/examples/endian_swapper/tests'
mkdir -p sim_build
cd ../cosim && make
make[4]: Entering directory '/home/travis/build/cocotb/cocotb/examples/endian_swapper/cosim'
gcc  -fPIC -pthread -fno-strict-aliasing -O2 -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 \
-fexceptions --param=ssp-buffer-size=4 -D_GNU_SOURCE \
-fwrapv -DNDEBUG -I/opt/python/3.8.0/include/python3.8 -c io.c -o io_module.o
gcc -pthread -shared  io_module.o -L/opt/python/3.8.0/lib  -o io_module.so
swig -python -outcurrentdir ../hal/endian_swapper_hal.h
make[4]: swig: Command not found
```

closed 2020-01-16 by eric-wieser

themperek, 2020-01-22:
Need to add swig here https://github.com/cocotb/cocotb/blob/master/Dockerfile#L9

eric-wieser, 2020-01-22:
@themperek: Mind making a PR?

eric-wieser, 2020-01-22:
Or shall I do it in 1314?

themperek, 2020-01-22:
5min

eric-wieser, 2020-01-22:
Done in #1316

commit 49985674 (2020-01-16): Install swig on linux ci (#1301)

Closes #1296, previously there was an error in the logs about `swig` not existing