cocotb/cocotb #1305: cocotb looking for python when only python3 exists

krupan, 2020-01-18:
I did a minimal Ubuntu 18.04 install which gave me a python3 but no python.  I then did a `pip3 install cocotb` and when I try to run a basic example I get this:

```
$ make
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make: python: Command not found
make results.xml
make[1]: Entering directory '/home/bryan/examples'
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
make[1]: python: Command not found
mkdir -p sim_build
/opt/msd-design-and-verification-suite-v1.0/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s counter -g2012   /home/bryan/examples/counter.sv
mkdir -p /home/bryan/examples/build/obj/x86_64
mkdir -p /home/bryan/examples/build/libs/x86_64
make -C /home/bryan/.local/lib/python3.6/site-packages/cocotb/share/lib/utils SIM=icarus
make[2]: Entering directory '/home/bryan/.local/lib/python3.6/site-packages/cocotb/share/lib/utils'
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
make[2]: python: Command not found
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DICARUS -c -I/home/bryan/.local/lib/python3.6/site-packages/cocotb/share/include -I  -o /home/bryan/examples/build/obj/x86_64/cocotb_utils.o cocotb_utils.c
gcc: error: /home/bryan/examples/build/obj/x86_64/cocotb_utils.o: No such file or directory
/home/bryan/.local/lib/python3.6/site-packages/cocotb/share/makefiles/Makefile.rules:57: recipe for target '/home/bryan/examples/build/obj/x86_64/cocotb_utils.o' failed
make[2]: *** [/home/bryan/examples/build/obj/x86_64/cocotb_utils.o] Error 1
make[2]: Leaving directory '/home/bryan/.local/lib/python3.6/site-packages/cocotb/share/lib/utils'
/home/bryan/.local/lib/python3.6/site-packages/cocotb/share/lib/Makefile:64: recipe for target '/home/bryan/examples/build/libs/x86_64/libcocotbutils.so' failed
make[1]: *** [/home/bryan/examples/build/libs/x86_64/libcocotbutils.so] Error 2
make[1]: Leaving directory '/home/bryan/examples'
/home/bryan/.local/lib/python3.6/site-packages/cocotb/share/makefiles/Makefile.sim:79: recipe for target 'sim' failed
make: *** [sim] Error 2
```

themperek, 2020-01-18:
Can you try `PYTHON_BIN=python3 make` ?

krupan, 2020-01-18:
That does fix it.  Now I wonder, should I just set that in my environment and move on?  Or will that break other things?

krupan, 2020-01-18:
Another question, for those that do have python (2) and python3 installed, is cocotb using both?  Are there problems with that?

themperek, 2020-01-18:
We will need https://github.com/cocotb/cocotb/pull/990 or similar to fix this. Long run hopefull will remove makefiles.

closed 2020-01-18 by themperek