cocotb/cocotb #1307: cocotb compile error or exception message

JaewonHur, 2020-01-21:
Recently, I'm working with verilator enabled cocotb, I got it from Antmicro (through git). 

But it seems that whenever I use Exception (or assert), and the program catches exception,
there is no exception message printed on the terminal.
Also, whenever there is compile error, I can not find backtrace which report which line causes error.

Is there any way to enable this?

Thank you.

themperek, 2020-01-21:
Verilator is supported directly from cocotb now (https://cocotb.readthedocs.io/en/latest/release_notes.html). Can you check with the latest cocotb and Verilator? 
What version of Verilator you are using.

JaewonHur, 2020-01-21:
I reinstalled cocotb,
and it works well.

Thank you.

closed 2020-01-21 by JaewonHur