cocotb/cocotb #1308: Does cocotb works fine for python version 2.7.6  ??

Vaibhavjain0901, 2020-01-21:
I am working on a shared host and python version installed is 2.7.6
I am using virtualenv for installing cocotb. While running endian_swapper example I am getting the below error:
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -Wstrict-prototypes -Waggregate-return -DFILTER -DVCS -shared -Xlinker -L/sw/packages/python/2.7.6/lib -o /auto/fpga-dump/vajain3/coco_cosim/cocotb/examples/endian_swapper/tests/build/libs/Linux64/libgpilog.so /auto/fpga-dump/vajain3/coco_cosim/cocotb/examples/endian_swapper/tests/build/obj/Linux64/gpi_logging.o   -lpython2.7 
/usr/bin/ld: /sw/packages/python/2.7.6/lib/libpython2.7.a(abstract.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/sw/packages/python/2.7.6/lib/libpython2.7.a: could not read symbols: Bad value
collect2: ld returned 1 exit status

Tool used is VCS

Is it related to compilation of python with --enable-shared switch?? any ideas??

thanks,
Vaibhav

themperek, 2020-01-21:
Version 1.3.0 works with python 2.7 as stated here: https://cocotb.readthedocs.io/en/v1.3.0/quickstart.html#pre-requisites

Maybe this can help: https://github.com/cocotb/cocotb/issues/1182#issuecomment-549515626

Please close the issue or change the title in case this is a different question.

imphil, 2020-01-21:
Did you compile with `--enable-shared`? It doesn't look like it; if not, you need to do that.

Vaibhavjain0901, 2020-01-22:
> Did you compile Python with `--enable-shared`? It doesn't look like it; if not, you need to do that.

Hi Phillip, I don't know whether python was compiled with '--enabled-shared' switch. Plus, I don't have permission to install python as I don't have root access.

eric-wieser, 2020-04-09:
This is definitely an `--enable-shared` issue. Opening #1616 to track producing a better error message.

Python 2.7 is no longer supported.

closed 2020-04-09 by eric-wieser