cocotb/cocotb #1313: Can we access multidimensional reg array using cocotb?

JaewonHur, 2020-01-22:
Is there any way to access multidimensional reg array in cocotb?
Such as 
```verilog
  reg[63:0] gpr[31:0];
```

Whenever I access the reg like this,
```python
  handle = dut.gpr
```

It reports 'RuntimeError: maximum recursion depth exceeded in cmp'

eric-wieser, 2020-01-22:
> It reports 'RuntimeError: maximum recursion depth exceeded in cmp'

Can you paste the full traceback?

JaewonHur, 2020-01-22:
Full traceback is below.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/cocotb/log.py", line 196, in _log_from_c
    function_name
  File "/usr/lib/python2.7/logging/__init__.py", line 1266, in makeRecord
    rv = LogRecord(name, level, fn, lno, msg, args, exc_info, func)
  File "/usr/lib/python2.7/logging/__init__.py", line 272, in __init__
    self.module = os.path.splitext(self.filename)[0]
  File "/usr/lib/python2.7/posixpath.py", line 98, in splitext
    return genericpath._splitext(p, sep, altsep, extsep)
  File "/usr/lib/python2.7/genericpath.py", line 109, in _splitext
    if p[filenameIndex] != extsep:
RuntimeError: maximum recursion depth exceeded in cmp
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/cocotb/log.py", line 196, in _log_from_c
Segmentation fault
```

eric-wieser, 2020-01-22:
Somethings gone very wrong there, you have both 2.7 and 3.6 paths in that traceback.

JaewonHur, 2020-01-23:
I solved the version problem, but still reporting similar error messages

eric-wieser, 2020-01-23:
It's really hard to see how that could be a recursion error, given how short the stack is. I suspect something is going wrong while trying to log a more useful error message.

eric-wieser, 2020-01-23:
Ah, I think I see what's happening now:

1. Something tries to log
2. The logger calls `get_sim_time` to try and get a time to show with the log message
3. `get_sim_time` fails, and `gpi` tries to log why it fails
4. The logger calls `get_sim_time` to try and get a time to show with the time that `get_sim_time` fails 
5. ...
```

eric-wieser, 2020-01-23:
Does the change in #1320 fix the `RecursionError`?

JaewonHur, 2020-01-23:
hm, I think it happens whenever I access multidimensional reg array.
I don't know why it tries to call ```get_sim_time```.

But with the simple test below, I get this error whenever I access ```dut.gpr```
(My cocotb version is 1.3.0)

```d_ff.v```
``` verilog
module d_ff(
  input clk,
  input d, 
  output reg q
);

reg[63:0] gpr[31:0];

always @(posedge clk)
begin
  q <= d;
end

endmodule
```

```test_d_ff.py```
``` python
import sys
import cocotb
from cocotb.regression import TestFactory
from cocotb.decorators import coroutine
from cocotb.triggers import Timer, RisingEdge

@coroutine
def clock_gen(clock, period=2):
    while True:
        clock <= 1
        yield Timer(period/2)
        clock <= 0
        yield Timer(period/2)
    
@coroutine
def run_test(dut):
    cocotb.fork(clock_gen(dut.clk))

    for i in range(100):
        yield RisingEdge(dut.clk)

    handle = dut.gpr   # Where error occurs

factory = TestFactory(run_test)
factory.generate_tests()
```

```Makefile```
```bash
# Verilator COCOTB Makefile

CWD=$(shell pwd)

SIM=verilator
CMD=verilator

TOPLEVEL_LANG ?= verilog

TOPLEVEL=d_ff
VERILOG_SOURCES=$(CWD)/$(TOPLEVEL).v

MODULE=test_$(TOPLEVEL)

CUSTOM_SIM_DEPS=$(CWD)/Makefile

include $(shell cocotb-config --makefiles)/Makefile.inc
include $(shell cocotb-config --makefiles)/Makefile.sim

sim: $(MODULE).py
```

eric-wieser, 2020-01-23:
> I don't know why it tries to call get_sim_time.

To include it in the log message

JaewonHur, 2020-01-23:
I tried. 
Still RecursionError, messages are below.

```
Traceback (most recent call last):
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/log.py", line 201, in _log_from_c
    function_name
  File "/usr/lib/python3.6/logging/__init__.py", line 1413, in makeRecord
    sinfo)
  File "/usr/lib/python3.6/logging/__init__.py", line 286, in __init__
    self.module = os.path.splitext(self.filename)[0]
  File "/usr/lib/python3.6/posixpath.py", line 129, in splitext
    return genericpath._splitext(p, sep, None, extsep)
  File "/usr/lib/python3.6/genericpath.py", line 130, in _splitext
    if dotIndex > sepIndex:
RecursionError: maximum recursion depth exceeded in comparison
Fatal Python error: Cannot recover from stack overflow.

Current thread 0x00007f08a43b9600 (most recent call first):
  File "/usr/lib/python3.6/genericpath.py", line 130 in _splitext
  File "/usr/lib/python3.6/posixpath.py", line 129 in splitext
  File "/usr/lib/python3.6/logging/__init__.py", line 286 in __init__
  File "/usr/lib/python3.6/logging/__init__.py", line 1413 in makeRecord
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/log.py", line 201 in _log_from_c
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/utils.py", line 68 in get_sim_time
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/log.py", line 112 in _format
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/log.py", line 179 in format
  File "/usr/lib/python3.6/logging/__init__.py", line 840 in format
  File "/usr/lib/python3.6/logging/__init__.py", line 994 in emit
  File "/usr/lib/python3.6/logging/__init__.py", line 865 in handle
  File "/usr/lib/python3.6/logging/__init__.py", line 1516 in callHandlers
  File "/usr/lib/python3.6/logging/__init__.py", line 1454 in handle
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/log.py", line 203 in _log_from_c
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/utils.py", line 68 in get_sim_time
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/log.py", line 112 in _format
  File "/home/jwhur/.local/lib/python3.6/site-packages/cocotb/log.py", line 179 in format
```

eric-wieser, 2020-01-23:
Ok, there's more going on here - what's actually happening is:

1. The comment I described above occurs, and python throws `RecursionError`.
2. `gpi_log` detects this failure, and tries to log it using `LOG_ERROR`
3. The python code is re-entered again.
4. ...

JaewonHur, 2020-01-23:
I temporarily solve it using decorator.
Using decorator to raise exception when the function is called recursively.

But it is incomplete.

eric-wieser, 2020-01-23:
Does #1321 solve the issue?

JaewonHur, 2020-01-24:
It still reports Segmentation fault.

eric-wieser, 2020-01-24:
Can you give the full log with #1321? Is the recursion issue gone at least?

JaewonHur, 2020-01-26:
Here's the log. 
```
        TESTCASE= TOPLEVEL=d_ff TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        sim_build/d_ff 
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on Verilator version 4.027 devel
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from /home/jwhur/.local/lib/python3.6/site-packages
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1580002541
     0.00ns INFO     cocotb.regression                         regression.py:196  in initialise                      Found test test_d_ff.run_test_001
     0.00ns INFO     cocotb.regression                         regression.py:360  in execute                         Running test 1/1: run_test_001
     0.00ns INFO     cocotb.test._my_test.0x7f6d6357e5c0       decorators.py:253  in _advance                        Starting test: "run_test_001"
                                                                                                                     Description: Automatically generated test
                                                                                                                     
                                                                                                                     
Segmentation fault
/home/jwhur/.local/lib/python3.6/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:59: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Error 139
make[1]: Leaving directory '/home/jwhur/Class/verilog_tutorial'
/home/jwhur/.local/lib/python3.6/site-packages/cocotb/share/makefiles/Makefile.sim:79: recipe for target 'sim' failed
make: *** [sim] Error 2
```

eric-wieser, 2020-01-26:
Did any new log messages appear after my fix, or not? It seems almost as though there's an error message that never quite makes it to stdout

JaewonHur, 2020-01-27:
It was all when the logging level was INFO. 
There was no additional log messages.

eric-wieser, 2020-01-27:
No faulthandler traceback any more either?

JaewonHur, 2020-01-27:
Yes. There was no traceback message either.

eric-wieser, 2020-01-27:
I wonder if perhaps the difference is caused by other changes in master between the version that was failing originally and the version you tried in my branch.

JaewonHur, 2020-01-28:
The difference is that it just asserts segmentation fault.

eric-wieser, 2020-02-02:
Frustrating. I guess the next thing to try is a debug build with gdb attached, so we can get a C stack trace.

cmarqu, 2020-02-02:
@JaewonHur See https://cocotb.readthedocs.io/en/latest/troubleshooting.html#attaching-a-debugger - we are using debug flags by default it seems.

lavanyajagan, 2020-02-15:
Could it be related to verilator ?
https://github.com/verilator/verilator/issues/2105

Have you tried running the same example with any other simulator ?.

eric-wieser, 2020-02-15:
This is almost certainly a verilator issue, but my worry was that we were crashing while trying to print out a verilator error message.

gretzteam, 2020-02-15:
I get the same error - and the same code works fine with xcelium. Must be verilator...

elgorwi, 2020-05-22:
@eric-wieser Just tried to run the example above and can confirm the RecursionError is still there. I've added a full log below.
I believe the issue is caused by: https://github.com/cocotb/cocotb/blob/bd4f5c4ef7827758c99a54ae8ad41f0f035c70a4/cocotb/share/lib/vpi/VpiImpl.cpp#L64

Looking at the ```vpi_get_time()``` implementation of verilator: https://github.com/verilator/verilator/blob/17e7da77f0bdcc60e2c5d9a16d0fd827e5479363/include/verilated_vpi.cpp#L1908
```vpi_get_time()``` does not clear the VPI error, so the subsequent call to ```vpi_get_error()``` will still see the previous uncleared error. 

We could fix the issue in the verilator repo by also clearing the VPI error in ```vpi_get_time()```. Looking at the rest of the VPI access functions in the verilator repo, vpi_get_time seems the only one where the VPI error is not cleared. That looks like an oversight.

Alternatively, we could fix it in the cocotb repo by inserting a dummy ```vpi_printf("")``` after ```vpi_get_time()```. The dummy VPI call will clear the VPI error and solve the problem as well.

I've added a second log below with either of the propsed fixes in place, which shows that the RecursionError is resolved.

I'm leaning towards fixing this in Verilator. What do you think @eric-wieser ?

Log without fix
```
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:245  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:168  in _initialise_testbench           Running on Verilator version 4.035 devel
     0.00ns INFO     cocotb                                      __init__.py:175  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from /Users/user/cocotb/cocotb
     0.00ns INFO     cocotb                                      __init__.py:195  in _initialise_testbench           Seeding Python random module with 1590121128
     0.00ns INFO     cocotb.regression                         regression.py:124  in __init__                        Found test test_dff.run_test_001
     0.00ns INFO     cocotb.regression                         regression.py:450  in _start_test                     Running test 1/1: run_test_001
     0.00ns INFO     ..otb.test.run_test_001.0x108eae290       decorators.py:256  in _advance                        Starting test: "run_test_001"
                                                                                                                     Description: Automatically generated test

     -.--ns ERROR    cocotb.gpi                         ..ib/vpi/VpiImpl.cpp:66   in get_sim_time                    VPI Error vpi_get_value: Unsupported format vpiIntVal
PROD Verilator
CODE (null)
FILE /usr/local/share/verilator/include/verilated_vpi.cpp:1603
     -.--ns ERROR    cocotb.gpi                         ..og/gpi_logging.cpp:248  in gpi_log_v                       Error calling Python logging function from C while logging the above
Traceback (most recent call last):
  File "/Users/user/cocotb/cocotb/log.py", line 249, in _filter_from_c
    return logging.getLogger(logger_name).isEnabledFor(level)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/logging/__init__.py", line 1935, in getLogger
    return Logger.manager.getLogger(name)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/logging/__init__.py", line 1230, in getLogger
    _acquireLock()
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/logging/__init__.py", line 219, in _acquireLock
    _lock.acquire()
RecursionError: maximum recursion depth exceeded while calling a Python object
     0.20ns ERROR    cocotb.gpi                                  VpiImpl.cpp:66   in get_sim_time                    VPI Error vpi_get_value: Unsupported format vpiIntVal
                                                                                                                     PROD Verilator
                                                                                                                     CODE (null)
                                                                                                                     FILE /usr/local/share/verilator/include/verilated_vpi.cpp:1603
     0.20ns ERROR    cocotb.gpi                                  VpiImpl.cpp:66   in get_sim_time                    VPI Error vpi_get_value: Unsupported format vpiIntVal
                                                                                                                     PROD Verilator
                                                                                                                     CODE (null)
                                                                                                                     FILE /usr/local/share/verilator/include/verilated_vpi.cpp:1603
     0.20ns ERROR    cocotb.gpi                                  VpiImpl.cpp:66   in get_sim_time                    VPI Error vpi_get_value: Unsupported format vpiIntVal
[... cut ...]
                                                                                                                     **                        SIM / REAL TIME : 5.95 NS/S                              **
                                                                                                                     *************************************************************************************

     0.20ns INFO     cocotb.regression                         regression.py:246  in tear_down                       Shutting down...
- :0: Verilog $finish
```

elgorwi, 2020-06-03:
@JaewonHur ,
This issue can now be closed. 
Verilator has just been updated with a PR that addresses array access in cocotb: https://github.com/verilator/verilator/pull/2395
To test this you will need to pull the latest Verilator version from git and compile/install manually or alternatively wait until the next (> v4.034) Verilator package release.

themperek, 2020-06-03:
I will close. Feel free to reopen if needed.

closed 2020-06-03 by themperek