cocotb/cocotb #1315: Signals name starting with underscore

JaewonHur, 2020-01-22:


eric-wieser, 2020-01-22:
I think this is by design - underscore names are reserved for cocotb internals.

Perhaps we should (or do) expose these as `dut.signals['_myname']` or `dut._.myname`

JaewonHur, 2020-01-23:
I see.

Thank you

eric-wieser, 2020-01-23:
Reopening, we should at least document this

cmarqu, 2020-04-11:
It works when you do it like this:
```
    # need to fill in _sub_handles
    _ = dir(dut)
    dut._sub_handles["_underscore_name"] <= 1
    yield Timer(1, 'ns')
    assert dut._sub_handles["_underscore_name"].value == 1
```

cmarqu, 2020-04-11:
Tried with Questa and Xcelium where it works, but unfortunately, Icarus doesn't see the signal.

eric-wieser, 2020-04-11:
Sounds related to #1610, where I found a similar problem with icarus

PR #2037 (2020-08-25): Add doc for accessing names starting w/ underscore
Closes #1315.
commit 1ded05fc (2020-08-25): Add doc for accessing names starting w/ underscore

Closes #1315.
closed 2020-08-25 by cmarqu