cocotb/cocotb #1323: Errors compiling VHPI code

cmarqu, 2020-01-24:
With #1137 in, I'm getting warnings-made-errors when compiling a VHDL design (shown here with gcc 9.2.0).
I have turned them back to be warnings to be able to show them all below:
```
[...]
g++ -fno-common -g -DDEBUG -fpic -std=c++11 -DVHPI_CHECKING -DIUS -Wall -Wextra -Wconversion -Wcast-qual -Wwrite-strings -Wnon-virtual-dtor -Woverloaded-virtual  -c -I/mypath/cocotb/cocotb/share/include -I/mypath/2019.10-python3.7/Linux-x86_64/include/python3.7m  -o /mypath/cocotb/tests/test_cases/test_external/build/obj/x86_64/VhpiImpl.o VhpiImpl.cpp
In file included from VhpiImpl.cpp:28:
VhpiImpl.h:210:5: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  210 |     const vhpiEnumT chr2vhpi(const char value);
      |     ^~~~~
VhpiImpl.h:202:17: warning: ‘virtual int VhpiSignalObjHdl::set_signal_value(double)’ was hidden [-Woverloaded-virtual]
  202 |     virtual int set_signal_value(const double value);
      |                 ^~~~~~~~~~~~~~~~
VhpiImpl.h:228:9: warning:   by ‘VhpiLogicSignalObjHdl::set_signal_value’ [-Woverloaded-virtual]
  228 |     int set_signal_value(std::string &value);
      |         ^~~~~~~~~~~~~~~~
VhpiImpl.cpp: In member function ‘virtual void VhpiImpl::get_sim_precision(int32_t*)’:
VhpiImpl.cpp:100:27: warning: conversion from ‘uint64_t’ {aka ‘long unsigned int’} to ‘double’ may change value [-Wconversion]
  100 |     double base = 1e-15 * femtoseconds;
      |                           ^~~~~~~~~~~~
g++ -fno-common -g -DDEBUG -fpic -std=c++11 -DVHPI_CHECKING -DIUS -Wall -Wextra -Wconversion -Wcast-qual -Wwrite-strings -Wnon-virtual-dtor -Woverloaded-virtual  -c -I/mypath/cocotb/cocotb/share/include -I/mypath/2019.10-python3.7/Linux-x86_64/include/python3.7m  -o /mypath/cocotb/tests/test_cases/test_external/build/obj/x86_64/VhpiCbHdl.o VhpiCbHdl.cpp
In file included from VhpiCbHdl.cpp:29:
VhpiImpl.h:210:5: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  210 |     const vhpiEnumT chr2vhpi(const char value);
      |     ^~~~~
VhpiImpl.h:202:17: warning: ‘virtual int VhpiSignalObjHdl::set_signal_value(double)’ was hidden [-Woverloaded-virtual]
  202 |     virtual int set_signal_value(const double value);
      |                 ^~~~~~~~~~~~~~~~
VhpiImpl.h:228:9: warning:   by ‘VhpiLogicSignalObjHdl::set_signal_value’ [-Woverloaded-virtual]
  228 |     int set_signal_value(std::string &value);
      |         ^~~~~~~~~~~~~~~~
VhpiCbHdl.cpp: In member function ‘virtual int VhpiSignalObjHdl::initialise(std::string&, std::string&)’:
VhpiCbHdl.cpp:271:63: warning: conversion from ‘long unsigned int’ to ‘int32_t’ {aka ‘int’} may change value [-Wconversion]
  271 |             m_value.bufSize = (m_num_elems)*sizeof(vhpiCharT) + 1;
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
VhpiCbHdl.cpp:293:60: warning: conversion from ‘long unsigned int’ to ‘int32_t’ {aka ‘int’} may change value [-Wconversion]
  293 |         m_binvalue.bufSize = m_num_elems*sizeof(vhpiCharT) + 1;
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
VhpiCbHdl.cpp: In member function ‘virtual int VhpiLogicSignalObjHdl::initialise(std::string&, std::string&)’:
VhpiCbHdl.cpp:341:38: warning: conversion from ‘long unsigned int’ to ‘int32_t’ {aka ‘int’} may change value [-Wconversion]
  341 |         m_value.bufSize = m_num_elems*sizeof(vhpiEnumT);
      |                           ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
VhpiCbHdl.cpp:353:60: warning: conversion from ‘long unsigned int’ to ‘int32_t’ {aka ‘int’} may change value [-Wconversion]
  353 |         m_binvalue.bufSize = m_num_elems*sizeof(vhpiCharT) + 1;
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
VhpiCbHdl.cpp: At global scope:
VhpiCbHdl.cpp:445:1: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  445 | const vhpiEnumT VhpiSignalObjHdl::chr2vhpi(const char value)
      | ^~~~~
VhpiCbHdl.cpp: In member function ‘virtual int VhpiSignalObjHdl::set_signal_value(long int)’:
VhpiCbHdl.cpp:553:62: warning: conversion from ‘long int’ to ‘vhpiEnumT’ {aka ‘unsigned int’} may change value [-Wconversion]
  553 |                 m_value.value.enumvs[m_num_elems-i-1] = value&(1L<<i);
      |                                                         ~~~~~^~~~~~~~
VhpiCbHdl.cpp:564:35: warning: conversion from ‘long int’ to ‘vhpiEnumT’ {aka ‘unsigned int’} may change value [-Wconversion]
  564 |             m_value.value.enumv = value;
      |                                   ^~~~~
VhpiCbHdl.cpp:569:34: warning: conversion from ‘long int’ to ‘vhpiIntT’ {aka ‘unsigned int’} may change value [-Wconversion]
  569 |             m_value.value.intg = value;
      |                                  ^~~~~
VhpiCbHdl.cpp:574:32: warning: conversion from ‘long int’ to ‘vhpiCharT’ {aka ‘char’} may change value [-Wconversion]
  574 |             m_value.value.ch = value;
      |                                ^~~~~
```
Ping @ktbarrett

cmarqu, 2020-01-24:
Actually, here are some more in the FLI part:

```
In file included from FliImpl.cpp:32:
FliImpl.h:258:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(std::string&)’ was hidden [-Woverloaded-virtual]
  258 |     virtual int set_signal_value(std::string &value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:294:9: warning:   by ‘virtual int FliEnumObjHdl::set_signal_value(long int)’ [-Woverloaded-virtual]
  294 |     int set_signal_value(const long value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:257:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(double)’ was hidden [-Woverloaded-virtual]
  257 |     virtual int set_signal_value(const double value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:294:9: warning:   by ‘virtual int FliEnumObjHdl::set_signal_value(long int)’ [-Woverloaded-virtual]
  294 |     int set_signal_value(const long value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:257:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(double)’ was hidden [-Woverloaded-virtual]
  257 |     virtual int set_signal_value(const double value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:336:9: warning:   by ‘FliLogicObjHdl::set_signal_value’ [-Woverloaded-virtual]
  336 |     int set_signal_value(std::string &value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:258:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(std::string&)’ was hidden [-Woverloaded-virtual]
  258 |     virtual int set_signal_value(std::string &value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:366:9: warning:   by ‘virtual int FliIntObjHdl::set_signal_value(long int)’ [-Woverloaded-virtual]
  366 |     int set_signal_value(const long value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:257:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(double)’ was hidden [-Woverloaded-virtual]
  257 |     virtual int set_signal_value(const double value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:366:9: warning:   by ‘virtual int FliIntObjHdl::set_signal_value(long int)’ [-Woverloaded-virtual]
  366 |     int set_signal_value(const long value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:258:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(std::string&)’ was hidden [-Woverloaded-virtual]
  258 |     virtual int set_signal_value(std::string &value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:392:9: warning:   by ‘virtual int FliRealObjHdl::set_signal_value(double)’ [-Woverloaded-virtual]
  392 |     int set_signal_value(const double value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:256:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(long int)’ was hidden [-Woverloaded-virtual]
  256 |     virtual int set_signal_value(const long value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:392:9: warning:   by ‘virtual int FliRealObjHdl::set_signal_value(double)’ [-Woverloaded-virtual]
  392 |     int set_signal_value(const double value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:257:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(double)’ was hidden [-Woverloaded-virtual]
  257 |     virtual int set_signal_value(const double value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:421:9: warning:   by ‘virtual int FliStringObjHdl::set_signal_value(std::string&)’ [-Woverloaded-virtual]
  421 |     int set_signal_value(std::string &value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.h:256:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(long int)’ was hidden [-Woverloaded-virtual]
  256 |     virtual int set_signal_value(const long value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:421:9: warning:   by ‘virtual int FliStringObjHdl::set_signal_value(std::string&)’ [-Woverloaded-virtual]
  421 |     int set_signal_value(std::string &value);
      |         ^~~~~~~~~~~~~~~~
FliImpl.cpp: In member function ‘virtual GpiObjHdl* FliImpl::native_check_create(void*, GpiObjHdl*)’:
FliImpl.cpp:221:67: warning: unused parameter ‘parent’ [-Wunused-parameter]
  221 | GpiObjHdl* FliImpl::native_check_create(void *raw_hdl, GpiObjHdl *parent)
      |                                                        ~~~~~~~~~~~^~~~~~
FliImpl.cpp: In member function ‘virtual const char* FliImpl::reason_to_string(int)’:
FliImpl.cpp:412:43: warning: unused parameter ‘reason’ [-Wunused-parameter]
  412 | const char *FliImpl::reason_to_string(int reason)
      |                                       ~~~~^~~~~~
In file included from FliCbHdl.cpp:28:
FliImpl.h:258:17: warning: ‘virtual int FliValueObjHdl::set_signal_value(std::string&)’ was hidden [-Woverloaded-virtual]
  258 |     virtual int set_signal_value(std::string &value);
      |                 ^~~~~~~~~~~~~~~~
[... 101 lines cut ...]
  256 |     virtual int set_signal_value(const long value);
      |                 ^~~~~~~~~~~~~~~~
FliImpl.h:421:9: warning:   by ‘virtual int FliStringObjHdl::set_signal_value(std::string&)’ [-Woverloaded-virtual]
  421 |     int set_signal_value(std::string &value);
      |         ^~~~~~~~~~~~~~~~
FliObjHdl.cpp: In member function ‘virtual int FliValueObjHdl::set_signal_value(long int)’:
FliObjHdl.cpp:142:49: warning: unused parameter ‘value’ [-Wunused-parameter]
  142 | int FliValueObjHdl::set_signal_value(const long value)
      |                                      ~~~~~~~~~~~^~~~~
FliObjHdl.cpp: In member function ‘virtual int FliValueObjHdl::set_signal_value(std::string&)’:
[... cut ...]
FliObjHdl.cpp:154:51: warning: unused parameter ‘value’ [-Wunused-parameter]
  154 | int FliValueObjHdl::set_signal_value(const double value)
      |                                      ~~~~~~~~~~~~~^~~~~
FliObjHdl.cpp: In member function ‘virtual int FliObjHdl::initialise(std::string&, std::string&)’:
FliObjHdl.cpp:80:25: warning: this statement may fall through [-Wimplicit-fallthrough=]
   80 |             m_indexable = true;
      |             ~~~~~~~~~~~~^~~~~~
FliObjHdl.cpp:81:9: note: here
   81 |         case GPI_MODULE:
      |         ^~~~
```

ktbarrett, 2020-01-25:
I don't have simulators available to me at home that I can test VHPI and FLI with. I kind of expected this.

One opinion is that `-Werror` in the makefiles for every build is horrible, it should be passed in by Travis for compliance checks ***only***. #1137 helps accomplish that -- but #1137 probably should have removed `-Werror`, my bad.

Another opinion is that the makefiles suck because there is no way to simply *compile* the libraries without the simulator present, due to how the makefiles are set up. Hopefully #1306 will fix this, but perhaps we should address this in the makefiles as well. At that point Travis should compile the libraries against every supported simulators for compliance checking.

#1137 was a kick in the rear.

cmarqu, 2020-01-25:
As a stopgap measure, #1328 removes the ``-Werror``.

ktbarrett, 2020-01-26:
Luckily, the warnings in VHPI I see could be changed without testing against a simulation (though I hope you can test it @cmarqu). However, the hidden virtual functions in the FLI with `double` and `long` able to do implicit conversions, as well as the fallthrough warning are troubling, and will require testing after fixes.

themperek, 2020-03-22:
@cmarqu Is this still a case?

cmarqu, 2020-03-22:
It wasn't last time I tried; if there are new such errors, we can open a new issue.

closed 2020-03-22 by cmarqu