cocotb/cocotb #1324: Helper function for marking expect_error/expect_fail?

cmarqu, 2020-01-24:
Right now, I have a need to express the following
```
@cocotb.test(expect_error=(
    lambda: cocotb.triggers.TriggerException if
    cocotb.SIM_NAME.startswith(("xmsim", "ncsim")) and cocotb.LANGUAGE in ["vhdl"] else False)())
```
Quite unwieldy, and I'll likely even need to add more simulator checks for the same test.
I'll need 17 such ``expect_error``s in our current ``tests/`` for Cadence simulators alone.
Should we have an (internal) helper function replacing the lambda (in e.g. ``utils.py``), or is there a better way?
This helper function could later become more capable and differentiate also between simulator _versions_ if needed (which will likely be the case with GHDL and Verilator in the future).

eric-wieser, 2020-01-24:
Why are you using a lambda, only to invoke it?

eric-wieser, 2020-01-24:
How about:
```
if cocotb.SIM_NAME.startswith(("xmsim", "ncsim")) and cocotb.LANGUAGE in ["vhdl"]:
    my_test = functoolss.partial(cocotb.test, expect_error=cocotb.triggers.TriggerException)
else:
    my_test = functools.partial(cocotb.test, expect_error=False)
@my_test
...
```

eric-wieser, 2020-01-24:
It would be good to get something like `@pytest.mark.xfail`, which can be stacked - but if we're going that direction, I'd rather just replace the `RegressionManager` with `pytest`...

cmarqu, 2020-01-24:
Right, the lambda was useless, but even without it, it's not exactly readable. The ``startswith`` part is using low-level information, and it's going to be duplicated a lot.
The ``my_test`` version looks even more like magic to the casual observer IMO.
I'm going the low-level for now and will reopen of the pain get too much when I go to different simulators.
Thanks for the ideas!

closed 2020-01-24 by cmarqu

eric-wieser, 2020-01-24:
Perhaps what we actually want is a`SimulatorInfo` object rather than just a bunch of strings?

cmarqu, 2020-01-24:
Oh, yes, that's a good idea!

cmarqu, 2020-01-26:
I see @mballance had the same general idea in https://github.com/mballance/cocotb/blob/tail-less/cocotb/env_info.py