cocotb/cocotb #1330: Enforcing compilation of libraries without warnings

ktbarrett, 2020-01-25:
#1137 added more warnings to the compilation process in an attempt to increase the C code quality and locate potential bugs (which it did and they were fixed in the common code and VPI). It also added the ability for the user to pass standardized compilation flags to the build process. Now that we are detecting more warnings, it's time we fix them, and once they are fixed, we should maintain the code quality with compliance checks.

We should add to travis compilation tests of all the libraries (save FLI which requires non-free headers) for every simulator with both GCC and Clang. The current makefiles were not made with this in mind so some work is necessary to add support.

eric-wieser, 2020-04-09:
Is there anything left to be done here? I'm not sure if we build with clang yet, but we do build everything but FLI in CI now.

Please close if you think there's nothing left to do.

themperek, 2020-04-09:
On osx clang is default as far as I am aware.

ktbarrett, 2020-04-09:
Do we build with `-Werror`? When I run tox locally, the build spits out warnings about C++ not supporting some warning flag.

themperek, 2020-04-09:
>Do we build with -Werror?

Yes, we do. It sees this is a different kind of warning.

themperek, 2020-04-16:
@ktbarrett Can we close this?

closed 2020-04-16 by ktbarrett