cocotb/cocotb #1331: Simulator "support"

ktbarrett, 2020-01-25:
The documentation lists many simulators as supported, and there are makefiles for many simulators, but the only simulator that currently passes all regression tests is Icarus. And this makes sense, we can't test changes against every simulators because they are not freely available to us for testing.

Should we evaluate whether we list a simulator as "supported", or perhaps categorize the support? We should also probably test changes more than we do for those non-free simulators. We could compile a list of people who are capable of testing certain simulators and require their sign-off for changes that would likely cause regressions with certain simulators. Thoughts?

I *might* be able to sign up for testing Questa and Riviera.

cmarqu, 2020-01-25:
We should at least tell the users how and with which OSes and simulator(s) we are testing (by pointing to Travis maybe). There used to be such an (outdated) information in the Introduction section of the docs.
Regarding testing more simulators, that's a goal of mine too. I actually recently started cleaning up the ``expect_errors`` for Cadence simulators; I'll continue with Synopsys VCS (I am seeing the same problem as #1261 here, will use the workaround initially).

ktbarrett, 2020-04-23:
I want some more discussion on this. How do we track and make known to users current issues with simulators? I think at a minimum we should introduce simulator specific tags to mark issues with.
@themperek @cmarqu @eric-wieser @imphil

cmarqu, 2020-04-23:
Simulator-specific labels could be a sensitive topic, given that apparently many license agreements forbid public comparisons with competitors etc. (I hear, I've never really read the legalese).

When, in "Simulator Support", we state that something is not supported, then we don't say why, so it could be our problem, not a simulator vendor's, so we are fine *there* IMHO.

ktbarrett, 2020-04-23:
Perhaps we need some legal support with that? But I'm not sure we are beholden to the license agreement because we aren't directly using their product.

cmarqu, 2020-05-01:
Coming from #875, maybe we could add very fine-grained testcases in the style of 
* read a single-bit reg
* write a single-bit reg
* read a vector reg
* write a vector reg
* read a single-bit generic/parameter

etc. that users can run themselves for their simulator?

ktbarrett, 2020-05-01:
Couldn't that just be the regressions? Currently we turn off certain tests because some simulators have bugs, or aren't feature complete. Within reason of course; I don't expect Icarus to all of a sudden support SystemVerilog, but I do expect Questa to not throw a fit on test_closedown.

Be no matter how we implement those tests, we have failing tests. Where/how do we keep track of which tests we know fail for each simulator?

eine, 2020-05-02:
@cmarqu, @ktbarrett, I think that you might find [VHDL/Compliance-Tests](https://github.com/VHDL/Compliance-Tests) to be inspiring for this task. The use case is different, as the purpose of Compliance-Tests is to track the support of VHDL features accross tools. However, issues such as not being able to test vendor tools on public CI, or to report certain test results, arise in both cases. Find a recent discussion at VHDL/Compliance-Tests#12.

From a technical point of view, it is proposed to use GitHub Actions in Compliance-Tests for several tasks that build an automatic tracking infrastructure. https://github.com/VHDL/Compliance-Tests/tree/LCS2019/issues is a Python script that allows to:

- A YAML files contains a list of tests/issues, including text description and, optionally, one or multiple MWEs.
- Any project maintainer can execute the script either locally or in a CI job, in order to have a new issue generated and labeled, for each item in the YAML file.
	- At any time, the script can be executed again. Existing issues will be updated. If some new test/feature was added to the YAML list, an issue will be created.
	- If the tests/features include MWEs and not only text descriptions, the MWEs can be written in the body of the issues following a certain structure (a markdown code block for each file).
- A separate tool can be used to extract the MWEs from GitHub issues, and have them executed. I proposed to use [eine/issue-runner](https://github.com/eine/issue-runner), which is a proof of concept that I wrote. However, it is possible to write any other tool/script for the same purpose.
    - Note that the idea of describing MWEs in issue bodies is primarily meant for user to write them manually. I.e., the existence of issue-runner is independent from having some YAML file where issues are generated/updated from.
	- issue-runner is composed of two parts: a golang tool and a GitHub Action (not a workflow, but a task). The golang tool extracts and runs the MWE. The Action (optionally) reports the results back. There are multiple mechanisms to report results:
		- Nothing, just go and check the log of the CI run.
		- Change the labels of the issue automatically. E.g. change from `triage` to `reproducible` or `fixed?`.
		- Upload artifacts from the CI run.
		- Edit the first message of the issue, or reply to the issue, in order to add either a reference to the CI run, or to copy some of the results. For instance, in project GHDL I'd like the `Bug occurred` block to be copied.
        - ...

This is expected to be used in any project that wants to track the support of a set of features (which are described in a YAML file), along with two GitHub Actions features:

- At least one scheduled Workflow which executes the Python script priodically, in order to keep the list of target issues up to date.
- At least one Step in a Workflow that is executed on `push` or `release` or when some issue is created/updated/edited, in order to report results.

Overall, my suggestion is for you to use GitHub Actions (it can be a complement to Travis, not a replacement) and to write your own action that interacts with the GitHub API. The Python script mentioned above, the golang tool and the JS action, all of them are working examples about how to do so. You pick your favourite language.

References to Compliance-Tests and issue-runner are for inspiration. Ideally, we would enhance them to reuse as much as possible in cocotb or other projects. Unfortunately, both are proofs of concepts and I cannot devote to move them to beta. Of course, I'm open to answering the questions you might have about Actions or GitHub's API (v3 or v4).

ktbarrett, 2025-02-21:
Between fixing many simulator-to-simulator inconsistencies and improving the simulator_support.rst in the documentation, I'd consider this done.

closed 2025-02-21 by ktbarrett