cocotb/cocotb #1345: Hosting cocotbext-spi module in cocotb github organization

Martoni, 2020-01-28:
I'm developping a spi bus cocotb module on my github profile : https://github.com/Martoni/cocotbext-spi

But it can be better to host it under the cocotb github "organization" no ?

Martoni, 2020-02-06:
I asked it because it's proposed in [documentation](https://github.com/cocotb/cocotb/blob/master/documentation/source/extensions.rst#code-hosting).
But I can see there is no extensions repository at the moment.

cmarqu, 2020-02-16:
Hi @sjalloq, I remember you also worked on SPI and mentioned you could share the code. Would you be willing to collaborate with Fabien on this extension?

sjalloq, 2020-02-16:
I thought I already had a repository with it in but that doesn't seem to be true.  I can upload it but it's not in the best state.

I've been using it recently and trying to work out the best way of structuring the code.  What I wanted was a low level base class that could be configured for frame length, timing and mode.  My driver and monitor use this class.  But what I wanted to do in my tests was subclass this to something that better matches my actual hardware and allow randomisation.  So I might have a 24-bit frame with 5-bit address and a 3-bit opcode.  What I was struggling with was how to convert to/from the base class when passing transactions from my monitor to my scoreboard.  That's probably just my inexperience in Python and verification.

I'll create a repo tomorrow and perhaps that might generate some ideas on the best implementation.  I'll also try to find time to review @Martoni 's code.

cmarqu, 2020-02-16:
I also recently had a need (but did not implement fully) for some specialties in SPI - Quad SPI (https://en.wikipedia.org/wiki/Serial_Peripheral_Interface#Quad_SPI), CRCs (different polynomials for address and data), and fill bits. I'm not sure if a single module handling all that can be written so that it is readable code, but maybe you can keep it in mind.
It would also be nice if there was enough useful output in debug mode, down to the bitlevel actually.

themperek, 2020-02-16:
One could potentially look around how UVM SPI (or similar) components look like.

imphil, 2020-02-16:
Since this extension is in very early stages and we have two potentially competing implementations: would the two of you mind finding common ground in one of the two repositories you're currently working on, and once things have slightly stabilized, we can move over the consolidated repository over to the cocotb namespace? 

Please also add a LICENSE, preferably a 2-clause BSD license for simplicity.

Martoni, 2020-02-17:
Hi @sjalloq 

Can you post the link to your project once you uploaded it ?

> would the two of you mind finding common ground in one of the two repositories you're currently working on, and once things have slightly stabilized, we can move over the consolidated repository over to the cocotb namespace?

I think we can found common ground yes. But I can't found @sjalloq code for the moment.

> Please also add a LICENSE, preferably a 2-clause BSD license for simplicity.

Ok I'll do that.

Martoni, 2020-02-17:
License [added](https://github.com/Martoni/cocotbext-spi/commit/ac0583f7b19eaa6cac295d248433248a896c56e1)

Martoni, 2020-02-20:
Hi @sjalloq,
Can you share your repository to compare our code please ?

Thank.

ktbarrett, 2025-08-27:
Considering the original library is now "obsolete" I think this can be closed.

closed 2025-08-27 by ktbarrett