cocotb/cocotb #1347: Two coroutines that fail on the same trigger are not handled gracefully by the scheduler.

benreynwar, 2020-01-28:
```
@cocotb.coroutine
async def coroutine1(timer):
    await timer
    assert False

@cocotb.coroutine
async def coroutine2(timer):
    await timer
    assert False

@cocotb.test()
async def test_top(dut):
    timer = triggers.Timer(100, units='ns')                                                                                            
    cocotb.fork(coroutine1(timer))                                                                                                     
    cocotb.fork(coroutine2(timer))
    await timer
```
This will fail when `_force_outcome` is called by the scheduler for the second time.  The scheduler needs to keep track of whether any coroutines that have been scheduled by this trigger have already failed.

ktbarrett, 2020-01-28:
This first on failing should immediately stop the test and kill any outstanding forked coroutines.

The problem logic is the last section in `Scheduler.unschedule`. It should set `self._terminate` and call `cleanup()` so the scheduler ends after handling the first failing forked coro.

eric-wieser, 2022-08-30:
Is there a similar problem if the two coroutines return different values?

marlonjames, 2022-08-30:
Only Exceptions cause `scheduler._abort_test()` to be called, so just returning a value shouldn't cause the same problem.

ktbarrett, 2022-08-30:
Sounds like a scheduler architecture issue. Once the test is unscheduled no more tasks should run which may also cause the test to fail.

aqeelmahroof, 2022-08-30:
Is it good to move this to a warning (or not to do this check), which will end the simulation clean
https://github.com/cocotb/cocotb/blob/997ee6dd906ea19954cb37f75a1406874b518db7/cocotb/scheduler.py#L1024 
```
Traceback (most recent call last):
  File "C:\Apps\WPy64-3940\python-3.9.4.amd64\lib\site-packages\cocotb\scheduler.py", line 357, in _react
    self._event_loop(trigger)
  File "C:\Apps\WPy64-3940\python-3.9.4.amd64\lib\site-packages\cocotb\scheduler.py", line 451, in _event_loop
    self._schedule(coro, trigger=trigger)
  File "C:\Apps\WPy64-3940\python-3.9.4.amd64\lib\site-packages\cocotb\scheduler.py", line 897, in _schedule
    self._unschedule(coroutine)
  File "C:\Apps\WPy64-3940\python-3.9.4.amd64\lib\site-packages\cocotb\scheduler.py", line 537, in _unschedule
    self._abort_test(e)
  File "C:\Apps\WPy64-3940\python-3.9.4.amd64\lib\site-packages\cocotb\scheduler.py", line 955, in _abort_test
    raise InternalError("Outcome already has a value, but is being set again.")
cocotb.scheduler.InternalError: Outcome already has a value, but is being set again.
```

This happens while running in middle of the regression tests . and terminates the rest of the tests (as in #3062). And some time questa sim will hang (but modelsim (10.4e) wont)  as well

ktbarrett, 2022-08-31:
That's not a good solution. After a test fails, no more tasks should run. If no more tasks run, no InternalError can happen. InternalError implies there is a bug in the implementation, so that's where it should be fixed.

ktbarrett, 2022-08-31:
#3062 shows that this affects `cocotb.start_soon`, so it is not necessarily due to the re-entrant behavior of `cocotb.fork`, so this is a legitimate scheduler bug that we might be able to fix.

imphil, 2022-08-31:
Well, there are two things: a bug in the scheduler implementation, and the way the user is informed about it. If we cannot find a quick solution to the scheduler bug (and we're clearly not, this bug has been open for > 2 years), it would at least be nice to have a workaround that is least annoying to our users when they encounter the bug. A crashing regression is certainly the most extreme measure we can take. But are there less extreme measures, such as a warning? I haven't looked into the details, but it sounds like that's feasible.

ktbarrett, 2022-08-31:
I discounted the issue because I thought the issue was due to `cocotb.fork`'s re-entrant behavior, making it unsolvable. It's the same issue, but it's kind of a different issue, so I wouldn't say it's 2 years old, there is new information since then. With `cocotb.start_soon`, this shouldn't be happening so I don't think we should be demoting it to a warning.

aqeelmahroof, 2022-09-01:
Thanks Guys!, its good to fix the issue without hiding it...

ktbarrett, 2022-09-02:
Seems the problem is that the event loop pops *all* coroutines waiting on a trigger off `triggers2coros` at once. This means other coroutines waiting on that trigger aren't killed in `cleanup`. Perhaps we need to move to a proper multimap semantics where we only pop one coroutine off the list at a time.

PR #3065 (2022-09-07): Fix test failure on same trigger
Closes #1347.
closed 2022-09-07 by ktbarrett