cocotb/cocotb #1348: 'kill' doesn't work on a coroutine when the coroutine has a ReadOnly trigger

benreynwar, 2020-01-29:
`counter` is a counter that logs a counter every clock cycle.
`test_intermed` forks a `counter` and then kills it after 3 clock cycles.
`test_top` awaits two `test_intermed` one after the other.

If everything is working properly the outputs from the two counters should not overlap, however when a `trigger.ReadOnly()` is present in the `counter` the `kill` does not work properly and the output from the first counter overlaps with the second.

```python
@cocotb.coroutine
async def clock(dut):
    while True:
        dut.clk <= 0
        await triggers.Timer(1, units='ns')
        dut.clk <= 1
        await triggers.Timer(1, units='ns')

@cocotb.coroutine
async def counter(dut, name):
    for clock_index in range(6):
        await triggers.RisingEdge(dut.clk)
        await triggers.ReadOnly()
        logger.warning('%s: %d', name, clock_index)

@cocotb.coroutine
async def test_intermed(dut, name):
    coro = cocotb.fork(counter(dut, name))
    for i in range(3):
        await triggers.RisingEdge(dut.clk)
    coro.kill()

@cocotb.test()
async def test_top(dut):
    cocotb.fork(clock(dut))
    await test_intermed(dut, 'a')
    await test_intermed(dut, 'b')
```

```
     1.00ns WARNING  debug2                                        debug2.py:25   in coroutine1                      a: 0               
     3.00ns WARNING  debug2                                        debug2.py:25   in coroutine1                      a: 1               
     5.00ns WARNING  debug2                                        debug2.py:25   in coroutine1                      a: 2               
     7.00ns WARNING  debug2                                        debug2.py:25   in coroutine1                      b: 0               
     7.00ns WARNING  debug2                                        debug2.py:25   in coroutine1                      a: 3              
     9.00ns WARNING  debug2                                        debug2.py:25   in coroutine1                      b: 1              
     9.00ns WARNING  debug2                                        debug2.py:25   in coroutine1                      a: 4              
```

benreynwar, 2020-01-29:
I think what's going on is that in the main `for coro in scheduling:` loop in the `_event_loop`, we don't consider that these coroutines might kill each other while we're iterating through them.

ktbarrett, 2020-01-29:
You diagnosis is correct. This only occurs if both coros are waiting on the same trigger and the coro that kills is run before the one that is killed.

eric-wieser, 2020-01-29:
Can we translate this to an example with asyncio.Event, and see what behavior occurs there? Would be nice to match semantics here, but the diagnosis sounds correct.

JaewonHur, 2020-02-19:
Is this issue resolved?

eric-wieser, 2020-04-24:
@ktbarrett, can you verify this was closed by #1684?

ktbarrett, 2020-04-24:
@eric-wieser It works now.

closed 2020-04-24 by ktbarrett