cocotb/cocotb #1350: coce coverage

Divyanirankari, 2020-01-29:
how can we find code coverage?

Divyanirankari, 2020-01-29:
for functional coverage we can use cocotb-coverage https://github.com/mciepluc/cocotb-coverage
but for code-coverage?

themperek, 2020-01-29:
You want to do code coverage of python testbench or RTL?

cmarqu, 2020-01-29:
See https://cocotb.readthedocs.io/en/latest/building.html#envvar-COVERAGE
If you want but don't get HDL code coverage with your simulator with this, you need to specify the simulator-specific extra arguments via https://cocotb.readthedocs.io/en/latest/building.html#var-EXTRA_ARGS

Divyanirankari, 2020-01-29:
> See https://cocotb.readthedocs.io/en/latest/building.html#envvar-COVERAGE
> If you want but don't get HDL code coverage with your simulator with this, you need to specify the simulator-specific extra arguments via https://cocotb.readthedocs.io/en/latest/building.html#var-EXTRA_ARGS

thanks

imphil, 2020-01-29:
That question has come up multiple times in the past. We should document the different types of "code coverage" (as discussed in this issue) and write a section on that in our docs.

Divyanirankari, 2020-01-29:
but how to see the percentage of code coverage

ktbarrett, 2021-01-08:
I feel like closing was a mistake. AFAIK there is no documentation on how to collect or view code coverage in either the CONTRIBUTING.md or the ReadTheDocs documentation.

Divyanirankari, 2021-01-08:
@ktbarrett 
we can find coverage using following python libraries:
1.  cocotb-coverage and https://github.com/JoseIuri/cocoTBCoverageView  
2.  pyvsc coverage libraries  and pyucis-viewer

ktbarrett, 2021-01-08:
So you were talking about HDL code coverage, not Python code coverage? If so we don't have documentation on that yet. We should point to those extensions at a minimum.

imphil, 2021-01-08:
Well, for HDL coverage, you only need extensions (such as cocotb-coverage) if you'd like to have (SystemVerilog-like) functional coverage, or want to drive your verificaton based on that. If you're only interested in things like line or toggle coverage, just use whatever functionality your simulator provides, cocotb is completely out of he picture here.

TomyPrudente, 2023-03-04:
Well, I'm using cocotb-test and iverilog. And I found an old tool called "covered" to do code coverage (HDL), but I couldn't make it work within the cocotb-test flow.
The interesting thing is that I couldn't find a good way to collect code coverage using cocotb-test and iverilog.
If I succeed I'll share it, just to have one example for the future.

ttDuong282, 2024-04-17:
I've been in trouble with displaying HDL code coverage (My design is Verilog code and my simulator is Iverilog), and I find this Issue. Is there anybody can find the way that can help getting HDL code coverage? I tried to do with "cocotb-coverage and https://github.com/JoseIuri/cocoTBCoverageView" but they do not work.

ktbarrett, 2025-08-27:
This issue is about HDL code coverage which is out of scope. We do list cocotb-coverage and pyvsc in [our wiki](https://github.com/cocotb/cocotb/wiki/Further-Resources). For Python code coverage we have `COCOTB_USER_COVERAGE` and `COVERAGE_RCFILE` documented. While cocotb Python coverage still deserves a proper tutorial, that isn't what's being asked here.

closed 2025-08-27 by ktbarrett