cocotb/cocotb #1360: Create new repository for cocotb bus implementations

imphil, 2020-01-31:
(This is an actionable follow-up to #1008.)

In our last maintainer call, @themperek, @eric-wieser and I agreed to try using a single cocotb-bus repository (naming TBD) for our "most important" bus implementations (e.g. the ones currently part of cocotb plus a couple other high-profile ones). This approach makes it easier to evolve the `Bus` class and related classes in lockstep with bus bus implementations. 

We'll give this approach a try and if it turns out to be infeasible for any reason we can always switch to more fine-grained repos in the future.

We can make use of the [CODEOWNERS](https://help.github.com/en/github/creating-cloning-and-archiving-repositories/about-code-owners) feature of GitHub to have maintainers assigned to subsets of the code base.

Any objections or suggestions? Otherwise I'll go ahead and do that, and move all existing bus implementations in cocotb over to this new repo.

eric-wieser, 2020-01-31:
I think we should perhaps finish up #1251 before doing that.

ktbarrett, 2021-01-30:
Done. cocotb-bus has been created.

closed 2021-01-30 by ktbarrett