cocotb/cocotb #1363: Intermittent simulator crash from UnicodeDecodeError

chiraag, 2020-02-02:
I am seeing an intermittent error at simulator start up when using `xcellium`

```
xrun(64): 18.03-s006: (c) Copyright 1995-2018 Cadence Design Systems, Inc.
Loading snapshot worklib.aes_gcm_top:sv .................... Done
xmsim: *W,DSEM2009: This SystemVerilog design is simulated as per IEEE 1800-2009 SystemVerilog simulation semantics. Use -disable_sem2009 option for turning off SV 2009 simulation semantics.
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
xcelium> source /cad/adi/apps/cadence/xlm/linux/18.03-s06/tools.lnx86/inca/files/xmsimrc
xcelium> run
UnicodeDecodeError: 'utf-8' codec can't decode bytes in position 0-1: invalid continuation byte
UnicodeDecodeError: 'utf-8' codec can't decode bytes in position 47-48: invalid continuation byte
     0.00ns ERROR    cocotb.gpi                                gpi_logging.c:224  in gpi_log                         Error calling Python logging function from C
```

Attaching the simulator log if it is any help.

ktbarrett, 2020-02-02:
Somewhere you are feeding non-UTF8 characters to a log call in C. It might be the path that you are storing cocotb in.

chiraag, 2020-02-02:
I was able to isolate the issue to one argument in the makefile:

```
EXTRA_ARGS := -f ../filelist.f
```

This even happens when the `filelist.f` file is completely empty.

eric-wieser, 2020-02-02:
Duplicate of #1234

chiraag, 2020-02-02:
Thanks! I did not notice the other issue. Is there a temporary workaround while the root cause is being addressed?

eric-wieser, 2020-02-02:
It's not entirely clear to me why you get an error about the failing log. #1321 may result in a marginally clearer error message.

Aside from that, we could insert the bytes into argv directly if decoding fails.

eric-wieser, 2020-02-02:
Mind running again on the latest master?

chiraag, 2020-02-02:
Will try this out now

eric-wieser, 2020-02-03:
Thanks. Guess we have two problems, not just one. In this code, not only are we failing, but our logging is also failing:

https://github.com/cocotb/cocotb/blob/573ed201933bf80ddbf1594ce196cc46f29f9730/cocotb/share/lib/embed/gpi_embed.c#L318-L324

eric-wieser, 2020-02-03:
@chiraag: is there any chance you have a support channel open with xcelium? It would be handy to get some feedback on what `vpi_get_vlog_info` is supposed to return for the `-f` argument, I haven't found any documentation suggesting it should do what it does.

eric-wieser, 2020-02-03:
Can you test if the patch in gh-1364 works for you, and paste the logs if it does?

chiraag, 2020-02-03:
This is the error log from [gh-1364](https://github.com/cocotb/cocotb/pull/1364)

```
xmsim: *W,DSEM2009: This SystemVerilog design is simulated as per IEEE 1800-2009 SystemVerilog simulation semantics. Use -disable_sem2009 option for turning off SV 2009 simulation semantics.
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:79   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
xcelium> source /cad/adi/apps/cadence/xlm/linux/18.03-s06/tools.lnx86/inca/files/xmsimrc
xcelium> run
     -.--ns WARNING  cocotb.gpi                                  gpi_embed.c:342  in embed_sim_init                  Unable to decode argv[4] = b'0\xe9\x84\x04' as utf8, leaving undecoded as bytes
     -.--ns ERROR    cocotb.gpi                                gpi_logging.c:244  in gpi_log_v                       Error calling Python logging function from C while logging the above
UnicodeDecodeError: 'utf-8' codec can't decode bytes in position 1-2: invalid continuation byte

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/tmp/cocotb_eric/cocotb/log.py", line 178, in _filter_from_c
    return logging.getLogger(logger_name).isEnabledFor(level)
  File "/tmp/anaconda3/envs/cocotb_dev/lib/python3.7/logging/__init__.py", line 1932, in getLogger
    return Logger.manager.getLogger(name)
  File "/tmp/anaconda3/envs/cocotb_dev/lib/python3.7/logging/__init__.py", line 1228, in getLogger
    if not isinstance(name, str):
SystemError: <built-in function isinstance> returned a result with an error set
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:372  in embed_sim_init                  Running on xmsim(64) version 18.03-s006
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:373  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from /tmp/cocotb_eric
Traceback (most recent call last):
  File "/tmp/cocotb_eric/cocotb/__init__.py", line 142, in _initialise_testbench
    process_plusargs()
  File "/tmp/cocotb_eric/cocotb/__init__.py", line 213, in process_plusargs
    if option.startswith('+'):
TypeError: startswith first arg must be bytes or a tuple of bytes, not str
     0.00ns ERROR    cocotb.gpi                                  gpi_embed.c:457  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:198  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
# check that the file was actually created, since we can't set an exit code from cocotb
test -f results.xml
make[1]: *** [results.xml] Error 1
```

chiraag, 2020-02-03:
Regarding support from Cadence. This is possible but I would like to understand the question regarding `vpi_get_vlog_info` and your current understanding of the bug a bit better before I open a support ticket.

Some questions I had:
- Can you give a high-level overview of the startup process that triggers the bug?
- What is the expected behavior from `vpi_get_vlog_info`?
- How do the other options from `EXTRA_ARGS` get processed?

eric-wieser, 2020-02-03:
> Can you give a high-level overview of the startup process that triggers the bug

Sure, it's essentially just "Invoke the simulator with an `-f <filename>` flag while loading a vpi module that calls `vpi_get_vlog_info`. `vpi_get_vlog_info` returns a struct with an `argv` member which is documented by the VPI specification as being an array of pointers to char:

[image]

Cadence appears to be either serving garbage, or has some undocumented feature where it hands us a pointer of some kind to some extra data where I'd expect to see the string `<filename>`.

If the former, then this is a bug in cadence. If the latter, then this is still a spec violation, so I'd like to at least understand what we're supposed to do with the `b'0\xe9\x84\x04'` value you got above.

> This is the error log

Thanks for testing - pleased to see the errors message works correctly, but looks like other parts are not expecting bytes instead of strings. I'll update that patch.

chiraag, 2020-02-03:
Thanks for the response. I understand this much better now.

Current flow of the `argv` string is:
`Makefile -> Simulator -> Python Interpreter (via vpi_get_vlog_info)`

In case there is no fix on the simulator forthcoming can there be a workaround where the Makefile generates a `tmpfile` with the simulator `argv` which Python reads directly?

eric-wieser, 2020-02-03:
Well, in reality we'll probably be just fine ignoring the mangled arguments, which was my intent in gh-1364. If we really cared about cocotb seeing the arguments in `<filename>`, then yes, we could do something like that. Alternatively, you could just run `sim $(cat my_args.f)` instead of `sim -f my_args.f`, which would also avoid the problem.

eric-wieser, 2020-02-04:
Mind trying #1366 and see if that fixes things?

If it does, could you add `print(sys.argv)` to the start of your test, and show me the output?

eric-wieser, 2020-02-07:
Ping @chiraag on the above

chiraag, 2020-02-07:
Sorry I got a bit busy during the week.
I tried this commit: [0627722](https://github.com/cocotb/cocotb/commit/06277225638f98fbc957f4b264645c6d4eabd7d8)

This is the output:
```
xmsim: *W,DSEM2009: This SystemVerilog design is simulated as per IEEE 1800-2009 SystemVerilog simulation semantics. Use -disable_sem2009 option for turning off SV 2009 simulation semantics.
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
xcelium> source /cad/adi/apps/cadence/xlm/linux/18.03-s06/tools.lnx86/inca/files/xmsimrc
xcelium> run
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on xmsim(64) version 18.03-s006
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:144  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from /tmp/cocotb_eric
     0.00ns INFO     cocotb                                      __init__.py:161  in _initialise_testbench           Seeding Python random module with 1581104758
     0.00ns INFO     cocotb.regression                         regression.py:196  in initialise                      Found test test_smp_dut.run_test
     0.00ns INFO     cocotb.regression                         regression.py:363  in execute                         Running test 1/1: run_test
     0.00ns INFO     cocotb.test.run_test.0x7fc6543e2390       decorators.py:259  in _advance                        Starting test: "run_test"
                                                                                                                     Description: Setup testbench and run a test.
     0.00ns INFO     cocotb.smp_dut_top                      test_smp_dut.py:60   in reset                           Resetting DUT
    15.00ns INFO     cocotb.smp_dut_top                      test_smp_dut.py:66   in reset                           Reset Complete
   205.00ns INFO     cocotb.smp_dut_top                      test_smp_dut.py:86   in run_test                        config complete
  3865.00ns INFO     cocotb.smp_dut_top                      test_smp_dut.py:96   in run_test                        test complete
  3965.00ns INFO     cocotb.regression                         regression.py:294  in handle_result                   Test Passed: run_test
  3965.00ns INFO     cocotb.regression                         regression.py:219  in tear_down                       Passed 1 tests (0 skipped)
  3965.00ns INFO     cocotb.regression                         regression.py:418  in _log_test_summary               *******************************************************************************
                                                                                                                     ** TEST                   PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                     *******************************************************************************
                                                                                                                     ** test_smp_dut.run_test    PASS         3965.00          0.16     25047.70  **
                                                                                                                     *******************************************************************************
                                                                                                                     
  3965.00ns INFO     cocotb.regression                         regression.py:435  in _log_sim_summary                *************************************************************************************
                                                                                                                     **                                 ERRORS : 0                                      **
                                                                                                                     *************************************************************************************
                                                                                                                     **                               SIM TIME : 3965.00 NS                             **
                                                                                                                     **                              REAL TIME : 0.19 S                                 **
                                                                                                                     **                        SIM / REAL TIME : 20942.84 NS/S                          **
                                                                                                                     *************************************************************************************
                                                                                                                     
  3965.00ns INFO     cocotb.regression                         regression.py:228  in tear_down                       Shutting down...
```

Everything seems to be working fine. Were you expecting more debug logging?

eric-wieser, 2020-02-07:
Any chance you could add `print(cocotb.argv)` to your testbench code, and show me what it prints? Otherwise, looks like it's fixed.

chiraag, 2020-02-07:
This is the print output:
```
['xrun', '-timescale', '1ns/1ps', '-f', '`ܓ\x03', '-licqueue', '-64', '-xmlibdirpath', 'sim_build', '-plinowarn', '-top', 'smp_dut_top', '-loadvpi', '/tmp/smp_dut/tests/build/libs/x86_64/libvpi', '-access', '+rwc', '-createdebugdb']
```

Yes I am good with solution and okay to close. Any idea when the next release will out with this fix.

eric-wieser, 2020-02-08:
Fantastic, thanks.

I don't know when a release will be out, but I'd hope the PR gets merged into master in the next week or so, and more speculatively the next release might happen in the next two or three months.

While it sounds like #1366 fixes this issue, it doesn't fix #1234 - so it would still be great if you could use your support channel to ask cadence what the mangled `-f` arguments mean in `vpi_get_vlog_info`.

PR #1366 (2020-02-12): Use PEP383 to embed non-decodable bytes in argv
Users who need the original bytes can use `.encode('utf8', 'surrogateescape')` to get them back.
Without this change, passing non-decodable bytes would just cause cocotb to fail to startup.

This behavior is consistent with how `sys.argv` is handled on UNIX platforms by python itself.

Alternative to gh-1364, and probably the better choice.

Fixes #1363

---
Workaround for #1234, using the same approach as is used for `sys.argv` in normal python.
closed 2020-02-12 by imphil