cocotb/cocotb #1365: Produce cocotb web site content

imphil, 2020-02-03:
cocotb currently has documentation and a GitHub page, but no web site. This situation makes it hard for us to promote cocotb to new or non-technical audiences. Furthermore, even for our long-time users, finding relevant documentation, status updates, etc. isn’t always easy. A web site could serve (at least) two purposes: present cocotb as a product to new users, and be the central information hub with a curated list of resources.

This issue is the starting point for a new web site. Before we think about presentation (layout), we first need to think about content. This is a bit more of an evolved exercise, since we at the same time need to refine the how we talk about cocotb in the first place.

**To collaboratively collect, refine and structure content I've created a [Google Doc](https://docs.google.com/document/d/1qbnCglW62AXUNnOo30Q1I7O249Q9jHJLluWbajN2TKc/edit?usp=sharing). Please open [this document](https://docs.google.com/document/d/1qbnCglW62AXUNnOo30Q1I7O249Q9jHJLluWbajN2TKc/edit?usp=sharing) and freely edit, comment, or suggest changes.**

imphil, 2020-02-03:
I'm also planning a kick-off brainstorming call for this one, please +1 on this message if you'd like to join and I'll start the process of finding a date/time.

I'd especially encourage all regular cocotb contributors and users to help out with the positioning of cocotb in the ecosystem. This will be essential if we want to increase the adoption of cocotb, and will help equally in working out what the "heart" of cocotb is. @ktbarrett @cocotb/maintainers

themperek, 2020-02-03:
Any chance to get any success stories (especially commercial)?

imphil, 2020-02-03:
> Any chance to get any success stories (especially commercial)?

I'd hope so. Please add that to the document, and include people or projects you know to which we can reach out.