cocotb/cocotb #1368: Build with -Wsuggest-override in CI runs

eric-wieser, 2020-02-04:
Split from #1149. Using `-Wsuggest-override` would catch some additional errors.
Unfortunately not all compilers support this flag, so we cannot just add it to our default set of build flags.

The attempt there was to use:
```
# version-specific flags
# easiest to compare versions with python
version_less_than_equal = $(shell $(PYTHON_BIN) -c "import sys; from distutils.version import LooseVersion; print(LooseVersion(sys.argv[1]) <= LooseVersion(sys.argv[2]))" $(1) $(2))
GXX_VERSION = $(shell g++ -dumpversion)
ifeq ($(call version_less_than_equal,4.9,$(GXX_VERSION)),True)
CXX_WARNS += -Wsuggest-override
endif
```
but this assumed that the user was using GCC.

We might be able to get away with just adding to `tox.ini`, assuming all our CI compilers are new enough.

imphil, 2020-02-04:
I'd simply pass it as environment variable which contains specific flags for CI runs. It's not uncommon to have more diagnostic output enabled in CI than in "normal" builds.

eric-wieser, 2020-02-04:
Let's have a go at this once #setuptools is merged

PR #3470 (2023-11-01): Compile dev builds with -Wsuggest-override
Fixes #1368
commit 72c400fa (2023-11-01): Compile dev builds with -Wsuggest-override

Fixes #1368
closed 2023-11-01 by imphil