cocotb/cocotb #1371: Weird dlopen errors in windows CI

eric-wieser, 2020-02-04:
Possibly new after #1369, cc @themperek.

https://travis-ci.org/cocotb/cocotb/jobs/646176059#L657
```
Unable to open lib Python.framework/Versions/3.7/Python (dlopen(Python.framework/Versions/3.7/Python, 9): image not found)
```

Seems to not actually prevent anything working, so maybe we just shouldn't be doing this at all.

themperek, 2020-02-05:
It is here:

https://github.com/cocotb/cocotb/blob/5571af7d5550dc7ca6144b3dc23df81b0b385e47/cocotb/share/lib/embed/gpi_embed.c#L115-L128

I have no idea what is this part for. We link `libpython` so what is the point here? It takes only warning and if you set `PYTHON_SO_LIB` to anything it seems to work.

imphil, 2023-11-01:
Probably not relevant any more, there are still tons of Windows issues, but this particular one is nothing we currently worry about.

closed 2023-11-01 by imphil