cocotb/cocotb #1372: Add support for MSVC

eric-wieser, 2020-02-06:
#1306 adds support for using setuptools, which knows how to invoke MSVC. However, initial experimentation revealed some follow-up work:

* Compiler flags need fixing (they are currently `-Werror` instead of `/W4` for instance)
* `.lib` files seem missing (https://travis-ci.org/themperek/cocotb/builds/646876123)

Opening this issue to track future work.

ktbarrett, 2020-02-10:
MSVC does not support C99 which is called out as our minimum supported C version. I don't know how trivial it would be to knock C support back to the C89 standard. Honestly, there does not need to be any C files in this project. Then standards are not an issue at all...

eric-wieser, 2020-02-10:
Numpy is able to build under MSVC and uses (most of) C99 - so I don't expect an issue there.

PR #1798 (2021-02-03): Add support for MSVC
This PR adds support for compiling with MSVC.

Depends on:
- [x] #1795 Don't include unistd.h on windows
- [x] #1797 Remove unused includes in embed.h
- [x] #1796 Add __declspec(dllexport) for functions/classes/variables to be exported

TODO:
- [x] Add CI
- [x] Clarify what warning level is desired on MSVC (at the moment the default `/W3` is used), W4 per recommendation on https://docs.microsoft.com/de-de/cpp/build/reference/compiler-option-warning-level
- [x] #2334 Portable compiler/linker flags
- [x] Make Makefiles work without lib prefix

Closes #1372, #2334, #2341
closed 2021-02-03 by ktbarrett