cocotb/cocotb #1374: Handling of string values on GPI handles

ktbarrett, 2020-02-07:
As discussed in #1256. 

In the `simulator` module, strings are parsed using the `"s"` format, which encodes the string using UTF-8 (which is default). This means the user cannot choose a representation for strings. Another issue is that users can't send certain sequences like those containing embedded NULs.

In the aforementioned issue, it was recommended this be changed to `"y"`. But if these GPI "set" functions require `bytes`, "get" should result in `bytes`. This is a change in behavior that will break existing simulations. Even with this change, it only offers some advantage because the GPI implementation cannot handle embedded NULs.

We can handle embedded NULs from the Python C API by using the parsing format `"y#"` which results in a `char *` and a `int` length. We could pass around both values, *or* `std::string` could handle this, as it is not a NUL-terminated string. But we can't use `std::string` in the simulator module or GPI because they are C files, not C++. There really isn't a reason for any source to be C and not C++.

I propose the following set of changes:
* Make ALL source files C++ files
* Update the GPI to use `std::string` to handle embedded NULs
* Update all the interfaces to handle embedded NULs
* Move parsing format of "get" and "set" value functions in the `simulator` module to `"y#"`
* Update `BinaryValue` to use `bytes` as it's representation.

eric-wieser, 2020-02-07:
> Update BinaryValue to use bytes as it's representation.

I don't think there's a particularly good reason to do this. Given the only allowable values are ascii (`01xz` etc), and this is text not bytes, the current `str` is just fine.

The problem comes with true `string` types, where the encoding is unspecified, and so we should refuse the temptation to guess an encoding.

> Update all the interfaces to handle embedded NULs

Do VPI and VHPI actually support these? Is the systemverilog string type allowed to contain these?

eric-wieser, 2020-02-07:
> But we can't use std::string in the simulator module or GPI because they are C files, not C++

However, we can easily pass through the pointer and length returned by `"#y"` to `gpi_set_...`, and then construct the `std::string` before calling into the C++ish API.

ktbarrett, 2020-02-08:
Just gauging what you thought was wrong, how far you wanted to go with it, and documenting current limitations.

eric-wieser, 2020-04-09:
As an update, I think item 4 on your list is done, and item 5 is rejected as by design. That leaves:

* Make ALL source files C++ files (aka, remove the C gpi interface and use only the C++ one)
* Update the GPI to use std::string to handle embedded NULs
* Update all the interfaces to handle embedded NULs (simulators may not support this anyway)

ktbarrett, 2020-04-09:
You are likely correct about simulators not supporting embedded NULs, and I'm not sure what the language specs on Verilog, SV, or VHDL say about them anyways, so we can ignore that. Also, I don't think moving to a C++ interface is a good idea (I'd eventually like to see other languages wrapping the GPI), especially if the only reason is to support embedded NULs (we can just add a string length or an end pointer to the interface).

closed 2020-04-09 by ktbarrett