cocotb/cocotb #1384: crv and coverage

Divyanirankari, 2020-02-14:


themperek, 2020-02-14:
@Divyanirankari Could you explain more?

Divyanirankari, 2020-02-14:
I have created constraint random generation and coverage for simple adder test . a and b and randomising. but how to get not repeated values. can you help me please?
https://github.com/Divyanirankari/cocotb_tests/tree/master/tests/adder_test_coverage/tests

themperek, 2020-02-14:
You should describe the problem better and rather ask here: https://github.com/mciepluc/cocotb-coverage

Divyanirankari, 2020-02-14:
@themperek 
i used randomise function  for adder inputs a and b ,but it is generating repeated values. and i dont know how to randomise non repeated values using cocotb_coverage.crv module in https://github.com/mciepluc/cocotb-coverage. 
otherwise i will ask here https://github.com/mciepluc/cocotb-coverage

Divyanirankari, 2020-02-14:
```
def adder_randomised_test(dut):
        """Test for adding 2 random numbers multiple times"""
        class SimpleRandomized(crv.Randomized):
               """   Constrained Random Instruction generation  """
               def __init__(self, A,B ):
                   crv.Randomized.__init__(self)
                   self.A= A
                   self.B= B
                   #defines random variables
                   # define mav_putvalue_instr as a random variable taking values from ins_list
                   self.add_rand("A",list(range(0,15))) 
                   self.add_rand("B",list(range(0,15))) 
                   c1 = lambda  A,B:A
                   c2 = lambda  A,B:B
                   #defining constraint for  instructions
                   self.add_constraint(c1)  
                   self.add_constraint(c2)  
        c=0 
        for i in range(16):
              # create randomized object instance
              x = SimpleRandomized(0,0)
              # randomize object with additional contraint
              #performs a randomization for all random variables meeting all defined constraints
              x.randomize() 
              print("mav_putvalue_instr constrained A , B ****** = 0x%X  0x%X" % (x.A,x.B))
              A = x.A  
              B = x.B  
              dut.A = A
              dut.B = B 
              c=c+1

              yield Timer(20, units='ns')

              if int(dut.X) != adder_model(A, B):
                    raise TestFailure(
                      "Randomised test failed with: %s + %s = %s" %
                      (int(dut.A), int(dut.B), int(dut.X)))
              else:  # these last two lines are not strictly necessary
                    dut._log.info("Ok!")

i using randomise method for adder inputs its generating repeated values. how to generate non repeated values?
```

themperek, 2020-02-14:
It does not look related to cocotb itself. Please close the issue here and ask https://github.com/mciepluc/cocotb-coverage

closed 2020-02-14 by Divyanirankari