cocotb/cocotb #1387: Use `build_clib` instead of `build_ext` for non python extension libraries

themperek, 2020-02-16:
From https://github.com/cocotb/cocotb/pull/1306#pullrequestreview-357334578

>Building normal libraries through the setuptools mechanisms for Python extension modules is highly confusing (as if the library situation in cocotb wasn't already confusing enough) and leads to ugly code, e.g. rewriting the file name in `build_ext`. Instead, use `build_clib` and/or a custom command in `setup.py`. As usual in Python packaging, it's mostly undocumented and a mess, but I've seen examples around that look reasonable.

themperek, 2020-02-16:
See also  #955

themperek, 2020-04-16:
@eric-wieser This seems to be not so easy taking account for build_clib state.  Can we close?

eric-wieser, 2020-04-16:
I found a stackoverflow post a couple days ago that might provide a path forward for this. I think we should leave it open.

eric-wieser, 2020-04-30:
xref #1739, which uses https://github.com/mdavidsaver/setuptools_dso instead