cocotb/cocotb #1392: [tracking] Unified VPI/VHPI/FLI library for all simulators

imphil, 2020-02-16:
**This is a tracking issue. File new issues for individual actionable work or investigation items, and link them here.**

Currently, we provide a "custom" libfli/libvpi/libvhpi for each supported simulator, even though they are almost identical (and should be, since they follow the same standard). Having only a single library for each supported interface would significantly ease the build and distribution process of cocotb, since we can remove one dimension from the build matrix (which already contains OS/potentially compiler/architecture). That doesn't mean that there cannot be any simulator-specific quirks. They only need to move from compile-time to runtime switches.

If the ultimate goal of a single libvpi/libfli/libvhpi isn't achievable, we should still work towards it by creating as few simulator-specific variants as possible.

- [ ] Remove IUS compile-time dependency in libvhpi: #1391
- [ ] Windows: Investigate runtime dynamic linking: #1390

enes377, 2020-02-20:
beatiful :))

themperek, 2020-02-20:
@imphil Not sure if this is possible. For example, VCS seems to use quite a different header file `vhpi_user.h ` (see https://github.com/cocotb/cocotb/issues/1103#issuecomment-546126426) if we ever to support VHDL.