cocotb/cocotb #1393: Document deprecated "out of the installation directory" or/and drop support.

themperek, 2020-02-16:
Drop support for deprecated "out of the installation directory", remove top-level makefiles and document the fact.
OR 
Document steps need to be compatible like `pip install` and set correctly `COCOTB` environment variable.

imphil, 2020-02-16:
I don't think we have a way to support using cocotb without some installation step any more, which isn't necessarily a bad thing.

So let's document it, e.g. like this:

> Cocotb must now be installed first before it can be used, please see the [installation instructions](https://cocotb.readthedocs.io/en/latest/quickstart.html#installing-cocotb) for details. The previously used method of setting the `COCOTB` environment variable to a cocotb source tree and including `$COCOTB/makefiles/Makefile.sim` is not supported any more.

To point people towards that fact, let's keep the two makefiles in `makefiles` for a release cycle longer, but remove all functionality and only let them output a error message with instructions how to install cocotb instead.

themperek, 2020-02-16:
Settings `COCOTB:=$(shell cocotb-config --share)` and including `$COCOTB/makefiles/Makefile.sim` should wok fine.

imphil, 2020-02-16:
How does that work without a pip install first?

themperek, 2020-02-16:
>How does that work without a pip install first?

No.  
But potentially this can avoid changing all Makefiles.

eric-wieser, 2020-04-29:
What is the state of this, @themperek?

themperek, 2020-04-29:
Can be closed after https://github.com/cocotb/cocotb/pull/1445

What is left is `remove top-level makefiles and` at some point.

closed 2020-04-29 by themperek