cocotb/cocotb #1402: SimFailure causing good tests to be marked as failed

ktbarrett, 2020-02-17:
#1382 is a stop-gap. It is actually recording tests that aren't being run that previously weren't, but there is still the issue of failing tests causing succeeding tests to fail if they cannot be tested because the simulator shut down.

Is this acceptable behavior? Does grouping tests into a single simulator run form an implicit contract that "it's okay to fail all of these tests if even one of them fails"? Should we make that more explicit with documentation or code changes to the `RegressionManager`? Or if that okay, meaning we should specifically state that it's okay to have DUT state carry from test to test?

imphil, 2020-02-17:
While the behavior isn't great, it's not uncommon to have such behavior in testing frameworks that somewhat sacrifice test isolation (in our case, a separate simulator run per test). And after your recent fix in #1382, it's safe behavior.

So for now, I'd go with documenting this fact and live with it, until someone wants to put more work into it (e.g. by restarting simulators, or providing an option to isolate test runs).

ktbarrett, 2023-12-09:
Instead of ending the regression with SimFailure if a regression must abort, we could instead mark the test as failing with `RegressionAborted` or similar exception class?