cocotb/cocotb #1404: Compilation error in ‘Tcl_Interp’ has no member named ‘result’

themperek, 2020-02-17:
There is an error on RHEL7 (gcc version 4.8.5 20150623 (Red Hat 4.8.5-36) (GCC)

```
 cocotb/share/lib/fli/FliCbHdl.cpp:164:44: error: ‘Tcl_Interp’ has no member named ‘result’
       unsigned long argc_u = strtoul(interp->result, NULL, 10);
                                              ^
  cocotb/share/lib/fli/FliCbHdl.cpp:183:43: error: ‘Tcl_Interp’ has no member named ‘result’
           argv[i] = new char[strlen(interp->result)+1];
                                             ^
  cocotb/share/lib/fli/FliCbHdl.cpp:184:33: error: ‘Tcl_Interp’ has no member named ‘result’
           strcpy(argv[i], interp->result);
                                   ^
```

Seems to be related to https://github.com/cocotb/cocotb/pull/1256  @ktbarrett

imphil, 2020-02-17:
Is cocotb picking up the right tcl library and header file from Questa, or is it using the system-provided one?

themperek, 2020-02-17:
It looks as if it is ok for Questa 10.x but 2019.x fails.

imphil, 2020-02-17:
Can you try reverting https://github.com/cocotb/cocotb/commit/b283b48368a3a7f507110c6699db3a6c7ce01677 and see if that fixes compilation? (If so, I'll back out this change until we find a solution that universally works.)

themperek, 2020-02-17:
Checked https://github.com/cocotb/cocotb/commit/b283b48368a3a7f507110c6699db3a6c7ce01677 seems to be a problem.

themperek, 2020-02-17:
The version of tcl changed from 8.5 to 8.6

imphil, 2020-02-17:
OK, revert in #1407. @ktbarrett can you look into finding a solution which works for both simulators? Ideally, one that doesn't involve conditional compilation (see #1392).

PR #1424 (2020-02-21): Implemented PLUSARGS for FLI
Fixed version of b283b48368a3a7f507110c6699db3a6c7ce01677. Uses `Tcl_GetStringResult` instead of accessing the internal `result` field, this works on TCL 8.5 and 8.6. Fixes #1404.

Finally got 2019.1 setup and tested this PR against it as well as 10.5c and 10.7c.

I tried using `Tcl_GetVar2Ex` to grab a reference to `argv`, then do processing on it, but there is something weird about how Questa maintains its TCL interpreter because it returns a message that the variable isn't defined. `mti_Cmd` seems to work.
closed 2020-02-21 by imphil