cocotb/cocotb #1422: Compile error in VhpiCbHdl.cpp: expected ‘)’ before ‘PRIu32’

cmarqu, 2020-02-18:
I'm getting the following error:
```
  In file included from cocotb/share/include/gpi.h:60,
                   from cocotb/share/lib/vhpi/../gpi/gpi_priv.h:32,
                   from cocotb/share/lib/vhpi/VhpiImpl.h:31,
                   from cocotb/share/lib/vhpi/VhpiCbHdl.cpp:29:
  cocotb/share/lib/vhpi/VhpiCbHdl.cpp: In member function ‘virtual GpiIterator::Status VhpiIterator::next_handle(std::string&, GpiObjHdl**, void**)’:
  cocotb/share/lib/vhpi/VhpiCbHdl.cpp:1150:69: error: expected ‘)’ before ‘PRIu32’
   1150 |         LOG_DEBUG("Unable to get the name for this object of type " PRIu32, type);
        |                                                                     ^~~~~~
  cocotb/share/include/gpi_logging.h:52:98: note: in definition of macro ‘LOG_DEBUG’
     52 | #define LOG_DEBUG(...)     gpi_log("cocotb.gpi", GPIDebug,         __FILE__, __func__, __LINE__, __VA_ARGS__);
        |                                                                                                  ^~~~~~~~~~~
  cocotb/share/include/gpi_logging.h:52:35: note: to match this ‘(’
     52 | #define LOG_DEBUG(...)     gpi_log("cocotb.gpi", GPIDebug,         __FILE__, __func__, __LINE__, __VA_ARGS__);
        |                                   ^
  cocotb/share/lib/vhpi/VhpiCbHdl.cpp:1150:9: note: in expansion of macro ‘LOG_DEBUG’
   1150 |         LOG_DEBUG("Unable to get the name for this object of type " PRIu32, type);
        |         ^~~~~~~~~
  error: command 'gcc' failed with exit status 1
```
This came in via #1361 (https://github.com/cocotb/cocotb/pull/1361/files#diff-8baf84ef545a81134ec388a5b271c5b6R1131).

eric-wieser, 2020-02-18:
Does adding `#include <cstdint>` fix this?

imphil, 2020-02-18:
On which simulator?

eric-wieser, 2020-02-18:
Moving `<cinttypes>` to be the first include may help here.

It's also possible that your compiler is broken, and does not conform with the standard. Can you give your compiler version?

cmarqu, 2020-02-18:
This is with both gcc 9.2.0 and 4.9.3.

eric-wieser, 2020-02-18:
Can you confirm that `--std=c++11` is being passed?

eric-wieser, 2020-02-18:
`__STDC_FORMAT_MACROS` is relevant here, but everything I found said that it is required to be set by default in C++11 mode.

cmarqu, 2020-02-18:
Yes, it is being passed, and also moving ``<cinttypes>`` to the top fixes it.

cmarqu, 2020-02-18:
I could open a PR, but there should be a good comment near the include line which I cannot write.

eric-wieser, 2020-02-19:
Can you try to bisect exactly which include `<cinttypes>` needs to be before? Eg, first reorder within:
```
#include <assert.h>
#include "VhpiImpl.h"
#include <limits>     // numeric_limits
#include <cinttypes>  // fixed-size int types and format strings
```
then assuming that `VhpiImpl.h` is the culprit, put `#include <cinttypes>` within the include stanzas of `VhpiImpl.h` until you find the offending header there, etc.

cmarqu, 2020-02-19:
Okay, I'm now at the point where it would need to go into ``cocotb/share/include/embed.h`` (as the 2nd include), but this is compiled as C, and so doesn't work.

cmarqu, 2020-02-19:
This sounds relevant (I'm on Redhat 6): https://github.com/MaskRay/ccls/issues/152

cmarqu, 2020-02-19:
Right, exporting ``CPPFLAGS "-D__STDC_FORMAT_MACROS"`` works, but I have to stop here for today.

eric-wieser, 2020-02-21:
> but this is compiled as C, and so doesn't work.

Put it in an `#ifdef __cplusplus` and you'll be able to continue bisecting.

cmarqu, 2020-02-21:
If I include``<cinttypes>`` before ``"inttypes.h"`` in
https://github.com/cocotb/cocotb/blob/master/cocotb/share/include/vpi_user.h#L40, then it works.

eric-wieser, 2020-02-21:
That was my hunch- I was trying to reproduce this on godbolt but could not.

My interpretation is that this is the problem:
```
// not required to include the macros - this is fine I think
#include <inttypes.h>
// include guard misfires, and forgets to define the macros which _are_ required
#include <cinttypes>
```

which is why swapping the order matters

Perhaps for that reason we should change `vhpi.h` to
```
// the inttypes.h header is contagiously broken on <the platform>
#ifdef __cplusplus
    #include <cinttypes>
#else
    #include <inttypes.h>
#endif
```
or something

cmarqu, 2020-02-21:
> Perhaps for that reason we should change `vhpi.h` to

I assume you mean ``vpi_user.h`` which is coming from the Verilog standard Annex G ("normative"), though this part is already modified so we don't have a pristine copy anyway.
Let me try that.

chiraag, 2020-03-06:
Will one of the open PRs fix this? Do you need anyone to test these out?

cmarqu, 2020-03-06:
It should be enough to export/setenv ``CPPFLAGS "-D__STDC_FORMAT_MACROS"`` and re-run pip (with ``-f`` or ``--force-reinstall``). If you do that, letting us know the result would be appreciated.

chiraag, 2020-03-07:
Which commit should I try this with?

cmarqu, 2020-03-07:
Any that you see this error message with.

chiraag, 2020-03-09:
I can confirm that this works for me. Is this the recommended fix going forward or will this be incorporated in the build scripts?

cmarqu, 2020-03-09:
Thanks for the confirmation. It will make its way into cocotb, we just haven't decided on the method to use yet.

PR #1430 (2020-03-16): Define __STDC_FORMAT_MACROS for old libc versions
The pre-C++11 glibc in e.g. Redhat 6 needs this symbol for the
cinttypes include.
Fixes #1422.
commit c4a573de (2020-03-16): Define __STDC_FORMAT_MACROS for old libc versions

The pre-C++11 glibc in e.g. Redhat 6 needs this symbol for the
cinttypes include.
Fixes #1422.
closed 2020-03-16 by cmarqu