cocotb/cocotb #1428: Put argv in `sys.argv` and not `cocotb.argv`

eric-wieser, 2020-02-20:
Right now, we set `sys.argv == ['cocotb']`, and put the simulator argv in `cocotb.argv`.

Is there a reason we don't just put the "real" argv in `cocotb.argv`?

Some things to check:
* What does pytest do?
* What do wsgi servers (bottle, cherrypy, gunicorn) do?

If we changed this, we could just set `cocotb.argv == sys.argv`.

ktbarrett, 2020-02-20:
* gunicorn passes arguments directly to the entry function, but gets these arguments from the entry point spec. https://stackoverflow.com/questions/8495367/using-additional-command-line-arguments-with-gunicorn
* pytest passes arguments directly to tests, with a bit of setup to expect certain arguments https://stackoverflow.com/questions/40880259/how-to-pass-arguments-in-pytest-by-command-line

eric-wieser, 2020-02-20:
My question is less "how are these passed" and more "what does sys.argv contain"?

ktbarrett, 2020-02-20:
pytest was a quick check, as expected it was the command line.

ktbarrett, 2020-02-20:
My thoughts are that we can set `sys.argv` to the command line, which we can now do on every interface, but I don't think `cocotb.argv` must be the same as `sys.argv`. It is conceivable that C (or something else) will pass additional arguments to the entry function, so that `cocotb.argv` might contain more, or just different, args. There is no reason to keep `cocotb.argv` around if it is synonymous with `sys.argv`.

eric-wieser, 2020-02-22:
`mod_wsgi` does the following:

https://github.com/GrahamDumpleton/mod_wsgi/blob/59f56107441f215f1af9ccc8cef82b6cd4b673d2/src/server/wsgi_interp.c#L571-L586

which is very similar to what we do today.

ktbarrett, 2020-03-01:
One more issue is that the `argv` available to cocotb is not the command's `argv` on all simulators. A recent test shows that on Icarus, `argv` does not represent the command's `argv`, simulator options are stripped first. On Questa, args are acquired from the outer `vsim` call, which does not have all the simulator args either. Putting this in `sys.argv` may give off the wrong impression.