cocotb/cocotb #1432: Can not get handle for some modules in hierarchy.

JaewonHur, 2020-02-21:
I'm using cocotb to test RISCV Rocket Chip, But I can not get handle for the csr instance.
The hierarchy is as follow.
```
Rocket Tile  (dut)
  -core
    -csr
    -bpu
    -alu
```
like this.

When trying to get handle like
```python
handle = dut.core.csr
```
cocotb reports
```
AttributeError: core contains no object named csr
```

but I can get handle for alu.
```
handle = dut.core.alu
```

Does anyone know reason for why I can access some module but some I can not?

themperek, 2020-02-21:
What simulator are you using? Did you try a different one? 

Cocotb just accesses whatever is available via vpi/vhpi/fli interface.

JaewonHur, 2020-02-21:
I'm using verilator.
I will reply again after trying with icarus verilog.

JaewonHur, 2020-02-22:
cocotb could not find handle for ``` csr ``` because, ``` csr ``` module vpi type was vpiScope.

Is there further plan to include vpiScope? 
or is there any way I can bypass?

themperek, 2020-02-23:
If you would show example Verilog code maybe someone would have an idea.

cmarqu, 2020-02-23:
Can you try to run cocotb's ``test_discovery`` on the design and see what shows up?

JaewonHur, 2020-02-24:
I tried it but I don't know whether it finds the
 ```csr``` instance.

Here's my result and test verilog file.
(change RocketTile.txt => RocketTile.v)

[test_discovery_result.txt](https://github.com/cocotb/cocotb/files/4242969/test_discovery_result.txt)
[RocketTile.txt](https://github.com/cocotb/cocotb/files/4242972/RocketTile.txt)

themperek, 2020-02-24:
I suggest to check with `Icarus` to me it looks like Verilator issues.

JaewonHur, 2020-03-05:
Yes, it is able to get handle when using ```Icarus```

It seems Verilator issue.

JaewonHur, 2020-04-04:
> I'm using cocotb to test RISCV Rocket Chip, But I can not get handle for the csr instance.
> The hierarchy is as follow.
>
> ```
> Rocket Tile  (dut)
>   -core
>     -csr
>     -bpu
>     -alu
> ```
>
> like this. [...]

closed 2020-11-24 by github-actions[bot]