cocotb/cocotb #1435: Tracing back through generators in pdb

toddstrader, 2020-02-21:
It's more of a general Python question, but I'm guessing it's a common issue for cocotb users. 
 Does anyone know of a way to use pdb to find the point at which a generator is created?  I often run into the problem where my code throws an exception and the call stack is useful until I get to this frame (I'm currently on an old version of cocotb):
```
   cocotb/cocotb/decorators.py(120)send()
-> return self._coro.send(value)
```
The problem often stems from some data which was passed into my generator (read: cocotb coroutine) when it was created.  But setting a breakpoint on the generator only gets me to the point where the generator has been entered.

If there's a strategy for tracing these things back, I think it would be worthwhile to add to the Troubleshooting section of the docs.  I'd be glad to add it if such a mechanism exists.

As a concrete (non-cocotb) example, if I have:
```python
#!/usr/bin/python3

def bar(val):
    yield 1
    return val

g = bar(7)
g.send(None)
```

And I run:
```
$ pdb3.6 ./bar.py
> bar.py(3)<module>()
-> def bar(val):
(Pdb) b bar
Breakpoint 1 at bar.py:3
(Pdb) c
> bar.py(4)bar()
-> yield 1
(Pdb) bt
  /usr/lib/python3.6/bdb.py(434)run()
-> exec(cmd, globals, locals)
  <string>(1)<module>()
  bar.py(8)<module>()
-> g.send(None)
> bar.py(4)bar()
-> yield 1
```

Now I see the send() which gets me into the generator, but I'd like to determine who created the generator in the first place (preferably conditionally on the parameters it was created with).  I'd like to be able to do this only knowing the generator and the parameters it was created with.

themperek, 2020-02-21:
Maybe this: https://cocotb.readthedocs.io/en/latest/building.html#envvar-COCOTB_PDB_ON_EXCEPTION ?

toddstrader, 2020-02-21:
> Maybe this: https://cocotb.readthedocs.io/en/latest/building.html#envvar-COCOTB_PDB_ON_EXCEPTION ?

That's handy, but I see where my exception is happening and can just pdb.set_trace() before that point.  My problem is that cocotb basically always leaves me running inside a generator.  And I'd like to be able to know where that generator was created.

eric-wieser, 2020-02-21:
Switch to using `async def` and `await` instead of `@coroutine` and `yield`, and this debugging will become much easier.

toddstrader, 2020-02-24:
The irony here is that I'm fighting bugs while migrating to Python 3.  So this presents a bit of a chicken/egg problem.  But as long as there's something better to look forward to I'll be happy.

Since I don't have any existing `async def` examples, I modified one of the cocotb examples:
https://github.com/toddstrader/cocotb/tree/async-coro-debug

Given this branch, I still see send() when breaking into the middle of a coroutine:
```python
> cocotb/cocotb/drivers/amba.py(157)write()
-> if int(result):
(Pdb) bt
  cocotb/cocotb/scheduler.py(346)react()
-> self._event_loop(trigger)
  cocotb/cocotb/scheduler.py(439)_event_loop()
-> self.schedule(coro, trigger=trigger)
  cocotb/cocotb/scheduler.py(748)schedule()
-> result = coroutine._advance(send_outcome)
  cocotb/cocotb/decorators.py(144)_advance()
-> return outcome.send(self._coro)
  cocotb/cocotb/outcomes.py(36)send()
-> return gen.send(self.value)
> cocotb/cocotb/drivers/amba.py(157)write()
-> if int(result):
(Pdb) 
```

I would instead (in addition to?) like to be able to discover that the `write()` was caused by line 45 of test_axi_lite_slave.py (in the write_address_0 test):
```python
    yield axim.write(ADDRESS, DATA)
```
I tried this with and without the coroutine decorators but the results were the same.

eric-wieser, 2020-02-24:
> I tried this with and without the coroutine decorators but the results were the same.

Unfortunately, you only get the benefit of `await` if you:
* Change the caller to use `await`
* Change the callee to use `async def`
* Remove the `coroutine` decorator from the caller.

For this reason, I intend to replace all `coroutine` functions within cocotb to use `async def` at some point :)

I think I now understand your problem - we implemented exception propagation through coroutines, but never implemented anything to help walk the stack from pdb.

toddstrader, 2020-02-24:
I see.  I've updated the above branch with what I believe is everything you've mentioned.  I now see the following when I run this example:

```
     0.00ns INFO     cocotb.regression                         regression.py:417  in execute                         Running test 1/5: write_address_0
     0.00ns INFO     ..st.write_address_0.0x7fde1e769240       decorators.py:253  in _advance                        Starting test: "write_address_0"
                                                                                                                     Description: Write to the register at address 0, verify the value has changed.
                                                                                                                     
                                                                                                                         Test ID: 0
                                                                                                                     
                                                                                                                         Expected Results:
                                                                                                                             The value read directly from the register is the same as the
                                                                                                                             value written.
                                                                                                                         
VCD info: dumpfile design.vcd opened for output.
> cocotb/cocotb/drivers/amba.py(156)write()
-> raise TypeError
(Pdb) bt
  cocotb/cocotb/scheduler.py(346)react()
-> self._event_loop(trigger)
  cocotb/cocotb/scheduler.py(439)_event_loop()
-> self.schedule(coro, trigger=trigger)
  cocotb/cocotb/scheduler.py(748)schedule()
-> result = coroutine._advance(send_outcome)
  cocotb/cocotb/decorators.py(257)_advance()
-> return super(RunningTest, self)._advance(outcome)
  cocotb/cocotb/decorators.py(144)_advance()
-> return outcome.send(self._coro)
  cocotb/cocotb/outcomes.py(36)send()
-> return gen.send(self.value)
  cocotb/examples/axi_lite_slave/tests/test_axi_lite_slave.py(45)write_address_0()
-> await axim.write(ADDRESS, DATA)
> cocotb/cocotb/drivers/amba.py(156)write()
-> raise TypeError
(Pdb) value
171
(Pdb) up
> cocotb/examples/axi_lite_slave/tests/test_axi_lite_slave.py(45)write_address_0()
-> await axim.write(ADDRESS, DATA)
(Pdb) DATA
171
(Pdb) c
   160.00ns ERROR    cocotb.regression                         regression.py:396  in _score_test                     Test Failed: write_address_0 (result was TypeError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "cocotb/examples/axi_lite_slave/tests/test_axi_lite_slave.py", line 45, in write_address_0
                                                                                                                         await axim.write(ADDRESS, DATA)
                                                                                                                       File "cocotb/cocotb/drivers/amba.py", line 156, in write
                                                                                                                         raise TypeError
                                                                                                                     TypeError
```

This is exactly what I was hoping for.  You seemed to indicate that there will be different information between when we see a stack trace from an exception vs when we are looking at the stack with pdb.  I do see that pdb observes the scheduler whereas the exception's stack trace does not.  But I see all the way up into the cocotb test in both cases, which may as well be the top of the stack as far as I'm concerned.

Is there some other subtle difference between these two modes of operation that I might care about in a more complicated example?  Or is this what I'm going to experience regardless of how much coroutine stacking/forking/etc. I do?

eric-wieser, 2020-02-24:
Whenever you `fork`, you end up with the disjointed stack like you had before.

The key thing to realize is that unless you do all three of the items I describe above, every `yield` or `await` becomes a `fork` (followed immediately by a join).

toddstrader, 2020-02-24:
OK, thanks for all the info.  I don't yet feel that I have enough of a handle on this to write anything in the docs (besides: try to use native coroutines and not generators), but I'm still experimenting so hopefully I'll get there.

I'll close this out for now and if I have any other examples around `fork` or whatnot I'll reopen or open a different issue.

closed 2020-02-24 by toddstrader