cocotb/cocotb #1443: Passing simulator options (flags) to Icarus Verilog

mballance, 2020-02-22:
Icarus Verilog makes a strong distinction between simulator options and simulation arguments. The first come before specification of the .vvp file, the later after. The cocotb Makefiles allow adding simulation arguments via the SIM_ARGS, EXTRA_ARGS, and PLUSARGS variables. However, they do not allow adding options to vvp itself.
Would having a SIM_FLAGS variable make sense? I'm envisioning something like the following for the VVP command in the simulation rule:

```
	PYTHONPATH=$(LIB_DIR):$(PWD):$(NEW_PYTHONPATH) $(LIB_LOAD) MODULE=$(MODULE) \
        TESTCASE=$(TESTCASE) TOPLEVEL=$(TOPLEVEL) TOPLEVEL_LANG=$(TOPLEVEL_LANG) COCOTB_SIM=1 \
        $(ICARUS_BIN_DIR)/vvp -M $(LIB_DIR) -m gpivpi $(SIM_FLAGS) $(SIM_BUILD)/sim.vvp $(SIM_ARGS) $(EXTRA_ARGS) $(PLUSARGS)
```

I've added $(SIM_FLAGS) reference directly after the gpivpi VPI module is loaded. 

Is there another way to do this? If not, would it make sense to have this capability?

themperek, 2020-02-22:
To me, it looks like `$(SIM_ARGS) $(EXTRA_ARGS)` should moved before `$(SIM_BUILD)/sim.vvp`  rather than adding new variables. What is the use of fo `$(SIM_ARGS)` at the current position?

mballance, 2020-02-22:
`$(SIM_ARGS)` and `$(EXTRA_ARGS)` allow non-plusarg arguments to be passed to VVP. Not entirely sure if this is useful, since it's not generally easy to get access to non-plusargs in simulation.

I think a fine solution would be to move `$(SIM_ARGS)` and `$(EXTRA_ARGS)` ahead of the .vvp file as you suggest. This would be consistent with cocotb's support for other simulators.

Should I create a pull request with this change?

themperek, 2020-02-22:
I also think this makes much more sense and how this now looks like a mistake.

It was added here: https://github.com/cocotb/cocotb/commit/bd579dc4f62810441bf2c6e9470df4e5549d4feb

closed 2020-04-23 by ktbarrett