cocotb/cocotb #1448: Simple example for README.md

themperek, 2020-02-23:
We once discussed that a very simple example would be more appropriate to be in the `README.md`.

Initially, I have come up with something like this (RTL from dff example).

```python
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge
from cocotb.result import TestFailure
import random

@cocotb.test()
async def simple_dff_test(dut):

    cocotb.fork(Clock(dut.c, 10).start(start_high=False))

    drive_data = [random.randint(0, 1) for r in range(10)]
    monitor_data = []

    for drive_val in drive_data:
        dut.d <= drive_val
        await RisingEdge(dut.c)
        monitor_data.append(dut.q.value)
        
    if drive_data[:-1] != monitor_data[1:]:
        raise TestFailure()

```

Open for discussion.

cmarqu, 2020-02-23:
``Clock`` should use the ``units`` parameter.
And maybe the RTL should be changed anyway to not require the ``start_high=False`"`?

themperek, 2020-02-23:
Other options:
```python
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge, Timer, Join
from cocotb.result import TestFailure
import random

async def bit_driver(event, signal, data):
    for drive_val in data:
        signal <= drive_val
        await event

async def bit_monitor(event, signal, cycles):
    data = []
    for _ in range(cycles):
        await event
        data.append(signal.value)
    return data

@cocotb.test()
async def simple_dff_test(dut):

    drive_data = [random.randint(0, 1) for r in range(10)]

    cocotb.fork(Clock(dut.c, 10, "us").start(start_high=False))
    cocotb.fork(bit_driver(RisingEdge(dut.c), dut.d, drive_data))
    monitor = cocotb.fork(bit_monitor(RisingEdge(dut.c), dut.d, len(drive_data)))

    await Join(monitor)

    if drive_data[:-1] != monitor.retval[1:]:
        raise TestFailure()
```

cmarqu, 2020-02-23:
Maybe we could show a progression of the same test toward a more abstracted test. (In your example, I would also add a timeout for the ``Join``. Oh, and docstrings :)).

themperek, 2020-02-23:
```
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge
import random

@cocotb.test()
async def simple_dff_test(dut):

    cocotb.fork(Clock(dut.c, 10, "us").start(start_high=False))

    for _ in range(10):
        dut.d <= random.randint(0, 1)
        await RisingEdge(dut.c)

```

eric-wieser, 2020-02-23:
Perhaps with an assert?

```python
import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge, FallingEdge

@cocotb.test()
async def test_simple_dff(dut):
    cocotb.fork(Clock(dut.c, 10, "us").start())

    for i in range(10):
        val = random.randint(0, 1)
        dut.d <= val
        await RisingEdge(dut.c)
        await FallingEdge(dut.c)
        assert dut.q == val, f"output was incorrect on the {i}th cycle"
```

themperek, 2020-02-24:
This also work. What do you think?

```python
import random
import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge

@cocotb.test()
async def test_simple_dff(dut):
    cocotb.fork(Clock(dut.c, 10, "us").start())

    for i in range(10):
        val = random.randint(0, 1)
        dut.d <= val
        await FallingEdge(dut.c)
        assert dut.q == val, f"output was incorrect on the {i}th cycle"
```

cmarqu, 2020-02-24:
We cannot use f-strings yet (only with 3.6+), and the import should say ``FallingEdge``.

themperek, 2020-02-25:
We could do something like "assert dut.q == val, "Oh no! The output is incorrect!"`

PR #1529 (2020-04-08): Update README.md with a simple example
Adding a very simple and easy to understand example to `README.md` that would better explain the idea behind cocotb. The result can be seen here: https://github.com/themperek/cocotb/tree/update_readmy

Contributions welcome.

Closes #1448
closed 2020-04-08 by themperek