cocotb/cocotb #1449: Add some basic assert methods ?

themperek, 2020-02-23:
```
assertEqual(a, b) 
assertNotEqual(a, b) 
assertTrue(x) 
assertFalse(x)
```

It would potentially make logging easier/better.

Based on https://docs.python.org/3/library/unittest.html#unittest.TestCase.debug

cmarqu, 2020-02-23:
I had actually started writing a check library based on https://vunit.github.io/check/user_guide.html which looks well thought out, but haven't gotten very far yet.

ktbarrett, 2020-02-23:
I like this solution better than using `assert`, so I think it is worthwhile.

Assert functions would allow for
1. delayed error message evaluation
2. throwing the proper exception: `TestFailure`

Right now we have support for `assert` statements that requires us to add it as another exception wherever `TestFailure` is expected. I think that was worked around by making `TestFailure` inherit from `AssertionError` as well as `TestComplete`. However, you still have to accept `AssertionError` and `TestComplete` if you want to accept any test ending exception. We can't work around that by making `TestComplete` inherit from `AssertionError`, that's not doing the right thing. The whole thing is just kind of messy.

eric-wieser, 2020-02-23:
Some questions here:

* How hard would it be to just use the unit-test ones directly, rather than writing our own?
* Is the eventual goal to be able to use pytest natively within the tests? (xref #pytest). If so, it might be better not to build these only to have them become defunct.

> However, you still have to catch AssertionError and TestComplete if you want to accept any test ending exception.

Are you describing `except` statements in cocotb internals, or in actual tests?

My gut feeling is that in the long run, having "test ending exception" objects is not a particularly good design. A pattern I've seen in pytest tests before is:
```python
def test_something():
    assert isinstance(a.foo(), Bar)
    try:
        assert a.foo().b == "b"
        assert a.foo().c == "c"
    except AssertionError as e:
        raise AssertionError("a.foo() was unexpected") from e
```
This approach doesn't work if `AssertionError` blows up the universe.

ktbarrett, 2020-02-23:
`AssertionError` only currently blows up the universe if it bubbles up to the root of a forked coroutine. This was done because the legacy semantics of `fork` are that. This could probably change, we discussed this in #922, but nothing came of it.

Nitori-, 2020-02-25:
I've had a lot of luck with [PyHamcrest](https://pyhamcrest.readthedocs.io/en/release-1.8/tutorial/) for adding richer asserts to Python packages (except when using pytest, which seems to use dark magic to get fancy looking asserts).

I will grant that it's a little verbose and I could see how some people might not like that.

```python
from hamcrest import *

@cocotb.test()
def the_test(dut):
    yield Timer(1)
    assert_that(2+2, equal_to(7))
```

```
Send raised exception:
Traceback (most recent call last):
  File "/nix/store/srfxgwcc5x53c08m3gfwxkxjmg9pwyz1-python3.7-cocotb-1.2.0/lib/python3.7/site-packages/cocotb/decorators.py", line 259, in _advance
    return outcome.send(self._coro)
  File "/nix/store/srfxgwcc5x53c08m3gfwxkxjmg9pwyz1-python3.7-cocotb-1.2.0/lib/python3.7/site-packages/cocotb/outcomes.py", line 37, in send
    return gen.send(self.value)
  File "/home/jwilcox/repo/sim/test.py", line 10, in the_test
    assert_that(2+2, equal_to(7))
  File "/nix/store/fplm7gph3m4ws74zvqby6gqbjl7czqy1-python3.7-PyHamcrest-1.9.0/lib/python3.7/site-packages/hamcrest/core/assert_that.py", line 43, in assert_that
    _assert_match(actual=arg1, matcher=arg2, reason=arg3)
  File "/nix/store/fplm7gph3m4ws74zvqby6gqbjl7czqy1-python3.7-PyHamcrest-1.9.0/lib/python3.7/site-packages/hamcrest/core/assert_that.py", line 57, in _assert_match
    raise AssertionError(description)
AssertionError:
Expected: <7>
     but: was <4>
```

themperek, 2020-02-25:
@Nitori- Nice find. If this work maybe we should document this and do not reinvent anything new for now.

PR #2028 (2021-01-15): Enable assertion rewriting with pytest
This means that code like `assert dut.signal.value == 3` will now show the actual signal value when the assertion fails.

I think by default this only applies to files called `test_*.py` or `*_test.py`.

Close #2027. Close #1449.
closed 2020-11-24 by github-actions[bot]