cocotb/cocotb #1451: VHPI compilation fails with Conda GCC

mballance, 2020-02-24:
I'm seeing a compilation failure on the VHPI library when using Conda and top-of-trunk cocotb. This is due to Conda using a fairly recent version of GCC (7.3.0) that only defines the standard format macros (eg PRIu32) for C code (not C++) or if a special macro is defined.
Definition of the macros is protected as follows in inttypes.h:

```
/* "C++ implementations should define these macros only when
 *  __STDC_FORMAT_MACROS is defined before <inttypes.h> is included."
 */
#if (! defined(__cplusplus)) || defined(__STDC_FORMAT_MACROS)
```

I have a PR that defines _STDC_FORMAT_MACROS in VhpiImpl.h, and this works for me.

ktbarrett, 2020-02-24:
Copy of #1422.

eric-wieser, 2020-04-09:
I think this is fixed, please open a new issue if you disagree.

closed 2020-04-09 by eric-wieser