cocotb/cocotb #1454: Getting 'max recursion error'' when trying to pickle dut object

gretzteam, 2020-02-24:
When trying to pickle the 'dut' object, pickle returns a 'max recursion error'.

gretzteam, 2020-02-24:
With full traceback:

131092.00ns ERROR    Test Failed: randmode (result was RecursionError)
                     Traceback (most recent call last):
                       File "/mnt/c/coco/design/tests/design.py", line 182, in randmode
                         plotoutput()
                       File "/mnt/c/coco/design/tests/design.py", line 187, in plotoutput
                         tb = pickle.load(f)
                       File "/home/miniconda3/lib/python3.7/site-packages/cocotb/handle.py", line 273, in __getattr__
                         return self._sub_handles[name]
                       File "/home/miniconda3/lib/python3.7/site-packages/cocotb/handle.py", line 273, in __getattr__
                         return self._sub_handles[name]
                       File "/home/miniconda3/lib/python3.7/site-packages/cocotb/handle.py", line 273, in __getattr__
                         return self._sub_handles[name]
                       [Previous line repeated 1491 more times]
                     RecursionError: maximum recursion depth exceeded

eric-wieser, 2020-02-24:
That's a fun one. It looks like `self._sub_handles` is hitting `__getattr__`, which then recurses!

ktbarrett, 2020-05-14:
@eric-wieser Did your change to `SimHandle.__getattr__` fix this? Is this even valid? Pickling a `SimHandle` seems like nonsense to me.

eric-wieser, 2020-05-14:
I have no idea whether i fixed it. I suspect my change to not store raw handles as integers means that python now refuses to pickle them at all.

eric-wieser, 2020-07-10:
I think we should leave this open, if picking duts is not in scope, it should raise a better error message.

themperek, 2020-07-10:
@gretzteam If I can ask what is the intension here (to pickle dut)?

closed 2025-10-17 by ktbarrett