cocotb/cocotb #1456: -Wno-unused-result error

tsengr0916, 2020-02-24:
I'm using the latest cocotb and gcc 5.3 version. When I install cocotb with pip3 install command, I got this error:
     gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -Icocotb/share/include -I/prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/coco3/inc/libs/blocks/pandace/sim/venv35/include -I/pkg/qct/software/python/3.5.2/include/python3.5m -c cocotb/share/lib/utils/cocotb_utils.c -o build/temp.linux-x86_64-3.5/cocotb/share/lib/utils/cocotb_utils.o -std=gnu99 -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wstrict-prototypes -Waggregate-return
    cc1: error: unrecognized command line option "-Wno-unused-result"
    error: command 'gcc' failed with exit status 1
    
    ----------------------------------------
Command "/prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/coco3/inc/libs/blocks/pandace/sim/venv35/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-hewyrayt-build/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-yht_goz3-record/install-record.txt --single-version-externally-managed --compile --install-headers /prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/coco3/inc/libs/blocks/pandace/sim/venv35/include/site/python3.5/cocotb" failed with error code 1 in /tmp/pip-hewyrayt-build/

How do I resolve this problem? Please advise.

Thanks,
Richard

eric-wieser, 2020-02-24:
Can you clarify exactly what command you ran, so we can work out what version you're using?

themperek, 2020-02-24:
This looks like `-Wno-unused-result` would come from `distutils` since it is at the beginning.

Can you paste here what you get with `python -m sysconfig`?

themperek, 2020-02-24:
From: https://github.com/atom/node-oniguruma/issues/25#issuecomment-69050611

It looks this option was added in GCC 4.4. 
http://repo.or.cz/w/tinycc.git/blobdiff/40a54c4399cc5e687d93add4b89bc16c19aaddfe..e79c3533ecefd004daad8a27333582b2dad7f0b0:/Makefile

So old version of gcc is not supported https://cocotb.readthedocs.io/en/latest/quickstart.html#pre-requisites

cmarqu, 2020-02-24:
Richard says he is using gcc 5.3. But maybe it's picking up an old version from somewhere else?

tsengr0916, 2020-02-27:
Thank you guys very much. That was my mistake. I thought I was using gcc 5.3, but actually my environment variable was not set correctly, and it picked up an older version.

After I added the gcc 5.3 executable at the front of the $PATH, it solves the issue. Sorry and thank you all for looking into this.

Richard

closed 2020-02-28 by cmarqu