cocotb/cocotb #1459: libcocotbvhpi.dll" library. The library does not exist or is corrupted

mao-karin, 2020-02-24:
I am working with pyhton 3.7, Aldec Riviera Pro 2017, cocotb 1.3 and MSYS2. When I run this test on jenkins and on remote PC I'm getting getting this issue. 

My log looks like this:

VHPI: Loading library 'C:/JenkinsSlave/workspace/Diceros_-_Regression_Tests_CoCoTB_mao/Vivado/diceros/ip_repo/registers_1.0/sim/build/libs/x86_64/libcocotbvhpi.dll'
VHPI: Cannot load the "C:/JenkinsSlave/workspace/Diceros_-_Regression_Tests_CoCoTB_mao/Vivado/diceros/ip_repo/registers_1.0/sim/build/libs/x86_64/libcocotbvhpi.dll" library. The library does not exist or is corrupted.

Solution tried so far: 
1. checked if the file is there and checked the dependencies of the dll file as well. (all good) 
2. Discovered Riviera has its own gcc version which is different from MSYS2 (mingw64) --not sure if that is a problem ? 
3. Played with PATH in environment variables (didn't work clearly)

Any suggestions will be helpful. I am really stuck at the moment. Thanks in advance!

cmarqu, 2020-02-25:
I fear this is a bug in cocotb 1.3.0 where a library renaming was not reflected in the Makefile for Riviera (and IUS): https://github.com/cocotb/cocotb/commit/14454b88b68327afb4e52dc5d6777aa99aa884fc#diff-5a5ae9f86b6f06a1263c126addca482fR87
Are you able to modify this Makefile (``cocotb/share/makefiles/simulators/Makefile.aldec``) in your installation?

@imphil Does this warrant a 1.3.1 release? :/

mao-karin, 2020-02-25:
Thank you! it's not complaining about the libcocotbvhpi.dll anymore after making the changes.
but the log file is coming up with this now:
make[1]: *** [/c/cocotb/cocotb/share/makefiles/simulators/Makefile.aldec:176: results.xml] Error 139
make: *** [/c/cocotb/makefiles/../cocotb/share/makefiles/Makefile.sim:80: sim] Error 2
starting test of:  registers
make results.xml

themperek, 2020-02-25:
If I manage to get a licanse will try to fix Aldec issues. It is very hard if one cannot test.

imphil, 2020-02-25:
We can of course do a 1.3.1 release with that fix.

cmarqu, 2020-02-25:
The error 139 may mean a segmentation fault (https://stackoverflow.com/questions/42882168/how-to-solve-exit-code-139-error-when-reading-from-file-on-unix#comment104094572_42890711), but I'm not sure if this helps any...

mao-karin, 2020-02-26:
I made the changes as suggested like this in the makefile.aldec. However, my DUT is in vhdl and since i didnt make any changes in the "else ifeq ($(TOPLEVEL_LANG),vhdl)" (not sure how this line works), does that matter?

`GPI_EXTRA:=
ifeq ($(TOPLEVEL_LANG),verilog)
    GPI_ARGS = -pli libvpi
ifneq ($(VHDL_SOURCES),)
    **GPI_EXTRA = cocotbvhpi**
endif

else ifeq ($(TOPLEVEL_LANG),vhdl)`

And i looked up the error in line 176 of the makefile.aldec (Error 139), do you think this happened because of the changes in the makefile or something else? :
[image]

cmarqu, 2020-02-26:
My guess is that the change in the Makefile now enables you to see the real problem. Is there some more information in ``sim_build/sim.log`` maybe?
(By the way, if you want to mark up multiple lines of code, use triple backticks before and after: ```)

themperek, 2020-02-26:
On windows, the situation is not so obvious. At some point, I had a license for a few days and fix this here 
 https://github.com/cocotb/cocotb/pull/1099 (can be done better).

mao-karin, 2020-02-27:
Unfortunately, I will have to get back to you next week with this due to some technical issues.

mao-karin, 2020-03-10:
Hi, sorry it took a while to get back to you all. I looked at the windows support that themperek has suggested. For a clean start, I reinstalled riviera, msys and cocotb. But I am running into the same issue. Thanks in advance!

This is the log:

# make TEST=registers_coco
make results.xml
make[1]: Entering directory '/c/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim'
set -o pipefail; cd sim_build && PYTHONPATH=/c/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/build/libs/x86_64:/c/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim:/C/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/../model:/c/cocotb:/C/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/../model: GPI_EXTRA= TOPLEVEL_LANG=vhdl \
PATH=/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/opt/bin:/c/Winpython64-3.7.4.1Zero/python-3.7.4.amd64:/c/Winpython64-3.7.4.1Zero/python-3.7.4.amd64/Scripts:/c/Winpython64-3.7.4.1Zero/python-3.7.4.amd64/libs:/c/Program\ Files\ \(x86\)/Common\ Files/Oracle/Java/javapath:/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0:/c/Windows/System32/OpenSSH:/c/Program\ Files/Amazon/cfn-bootstrap:/c/Program\ Files\ \(x86\)/GitExtensions:/c/Program\ Files/Git/cmd:/c/Xilinx/Vivado/2017.3/bin:/c/Xilinx/Vivado/2017.3/lib/win64.o:/c/Xilinx/SDK/2017.3/bin:/c/Xilinx/SDK/2017.3/gnu/microblaze/nt/bin:/c/Xilinx/SDK/2017.3/gnu/arm/nt/bin:/c/Xilinx/SDK/2017.3/gnu/microblaze/linux_toolchain/nt64_le/bin:/c/Xilinx/SDK/2017.3/gnu/aarch32/nt/gcc-arm-linux-gnueabi/bin:/c/Xilinx/SDK/2017.3/gnu/aarch32/nt/gcc-arm-none-eabi/bin:/c/Xilinx/SDK/2017.3/gnu/aarch64/nt/aarch64-linux/bin:/c/Xilinx/SDK/2017.3/gnu/aarch64/nt/aarch64-none/bin:/c/Xilinx/SDK/2017.3/gnu/armr5/nt/gcc-arm-none-eabi/bin:/c/Xilinx/SDK/2017.3/tps/win64/cmake-3.3.2/bin:/c/Aldec/Riviera-PRO-2017.10-x64/bin:/c/Aldec/Riviera-PRO-2017.10-x64/mingw/bin:/mingw64/bin:/mingw64/lib:/usr/bin:/usr/include:/:/c/Aldec/ALINT-PRO-2017.01-x64:/c/Aldec/ALINT-PRO-2017.01-x64/bin:/c/Users/Administrator/AppData/Local/Microsoft/WindowsApps:/c/Users/Administrator/AppData/Local/Programs/Microsoft\ VS\ Code/bin:/c/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/build/libs/x86_64 MODULE=registers_coco TESTCASE= TOPLEVEL=Sim_Wrapper COCOTB_SIM=1 /c/Aldec/Riviera-PRO-2017.10-x64/bin/vsimsa -do runsim.tcl | tee sim.log
/bin/sh: line 1:  1031 Segmentation fault      PYTHONPATH=/c/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/build/libs/x86_64:/c/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim:/C/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/../model:/c/cocotb:/C/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/../model: GPI_EXTRA= TOPLEVEL_LANG=vhdl PATH=/mingw64/bin:/usr/local/bin:/usr/bin:/bin:/opt/bin:/c/Winpython64-3.7.4.1Zero/python-3.7.4.amd64:/c/Winpython64-3.7.4.1Zero/python-3.7.4.amd64/Scripts:/c/Winpython64-3.7.4.1Zero/python-3.7.4.amd64/libs:/c/Program\ Files\ \(x86\)/Common\ Files/Oracle/Java/javapath:/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0:/c/Windows/System32/OpenSSH:/c/Program\ Files/Amazon/cfn-bootstrap:/c/Program\ Files\ \(x86\)/GitExtensions:/c/Program\ Files/Git/cmd:/c/Xilinx/Vivado/2017.3/bin:/c/Xilinx/Vivado/2017.3/lib/win64.o:/c/Xilinx/SDK/2017.3/bin:/c/Xilinx/SDK/2017.3/gnu/microblaze/nt/bin:/c/Xilinx/SDK/2017.3/gnu/arm/nt/bin:/c/Xilinx/SDK/2017.3/gnu/microblaze/linux_toolchain/nt64_le/bin:/c/Xilinx/SDK/2017.3/gnu/aarch32/nt/gcc-arm-linux-gnueabi/bin:/c/Xilinx/SDK/2017.3/gnu/aarch32/nt/gcc-arm-none-eabi/bin:/c/Xilinx/SDK/2017.3/gnu/aarch64/nt/aarch64-linux/bin:/c/Xilinx/SDK/2017.3/gnu/aarch64/nt/aarch64-none/bin:/c/Xilinx/SDK/2017.3/gnu/armr5/nt/gcc-arm-none-eabi/bin:/c/Xilinx/SDK/2017.3/tps/win64/cmake-3.3.2/bin:/c/Aldec/Riviera-PRO-2017.10-x64/bin:/c/Aldec/Riviera-PRO-2017.10-x64/mingw/bin:/mingw64/bin:/mingw64/lib:/usr/bin:/usr/include:/:/c/Aldec/ALINT-PRO-2017.01-x64:/c/Aldec/ALINT-PRO-2017.01-x64/bin:/c/Users/Administrator/AppData/Local/Microsoft/WindowsApps:/c/Users/Administrator/AppData/Local/Programs/Microsoft\ VS\ Code/bin:/c/JenkinsSlave/workspace/Vivado/diceros/ip_repo/registers_1.0/sim/build/libs/x86_64 MODULE=registers_coco TESTCASE= TOPLEVEL=Sim_Wrapper COCOTB_SIM=1 /c/Aldec/Riviera-PRO-2017.10-x64/bin/vsimsa -do runsim.tcl
      1032 Done                    | tee sim.log
make[1]: *** [/c/cocotb/cocotb/share/makefiles/simulators/Makefile.aldec:176: results.xml] Error 139
make[1]: Leaving directory '/c/JenkinsSlave/workspace/Diceros_-_Regression_Tests_CoCoTB_mao/Vivado/diceros/ip_repo/registers_1.0/sim'
make: *** [/c/cocotb/makefiles/../cocotb/share/makefiles/Makefile.sim:80: sim] Error 2

themperek, 2020-03-10:
@mao-karin It will be hard to fix all this for version 1.3 at this point. Your environment seems also very complicated. What you can try is to use this (1.4.dev): https://github.com/cocotb/cocotb/pull/1462

`pip install https://github.com/themperek/cocotb/archive/better_windows2.zip`

mao-karin, 2020-03-11:
@themperek downloaded 1.4dev and looked at #1462 too.

1. I cant pip install  pip install https://github.com/themperek/cocotb/archive/better_windows2.zip
- ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly
- So if I just replace the 1.4dev files with your files will that work as an alternative?

2. in #1462 could you be a little more clear with "in embedded python interpreter in Makefile.pylib.Msys". Does that mean setting PYTHONHOME to the msys python path (msys comes with python 3.8) or the windows python path (py 3.7)?

3. in #1462 could you also please clarify "set PATH to include a path to cocotb compiled libraries in Makefile.pylib.Msys"
Does this mean "C:\cocotb\cocotb\share\lib" ?

4. Also in the in #1462, i think it would be helpful to add install "pacman -S make" with the msys instruction you've written.

mao-karin, 2020-03-14:
Hi guys can I get some help regarding this please? Still stuck at this :(
@themperek @cmarqu @imphil

themperek, 2020-03-14:
@mao-karin You seem to have very complicated environment with multiple installations GCC and possibly other (from Xilinx, Aldec,...). They are all possibly not compatible with each other. 

>ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly

You seem to have something very wrong with your installation.

>in #1462 could you be a little more clear with "in embedded python interpreter in Makefile.pylib.Msys". Does that mean setting PYTHONHOME to the msys python path (msys comes with python 3.8) or the windows python path (py 3.7)?
in #1462 could you also please clarify "set PATH to include a path to cocotb compiled libraries in Makefile.pylib.Msys"
Does this mean "C:\cocotb\cocotb\share\lib" ?

You should not have to set anything this is information for ccootb developers.

themperek, 2020-04-16:
In the current development version, this part has significant changes and should be better. 
Please also check https://docs.cocotb.org/ also related to Aldec.

Please reopen if the problem is still there.

mao-karin, 2020-05-05:
@binupr

mkzo7, 2020-05-07:
Hello,
I'm continuing Mao's effort regarding cocotb vs Python 3 vs Riviera PRO compatibility.
I'd like to try the recent cocotb dev version whether it solves our problems, but I'm having problems installing it, when using this command:
$ pip install https://github.com/themperek/cocotb/archive/better_windows2.zip
It complains about:
 error: Microsoft Visual C++ 14.0 is required
I installed Visual C++ 2019, but it fails pip install with a different problem.
Do I need Visual C++ 14? If yes where can I find it?
Cheers,
Marcin

themperek, 2020-05-07:
You hve to use `pip install --global-option build_ext --global-option --compiler=mingw32 https://github.com/cocotb/cocotb/archive/master.zip`  (do not use `better_windows2.zip`) more info: https://stackoverflow.com/questions/3297254/how-to-use-mingws-gcc-compiler-when-installing-python-package-using-pip

Updated documentation is on the way.

themperek, 2020-05-07:
I will close this since the installation process has significantly changed. Please reopen if the problem is still there.

PR #1472 (2020-03-15): Fix wrong vhpi library name for ius and aldec in v1.3
Resolves #1459
closed 2020-05-07 by themperek