cocotb/cocotb #1460: Unused pieces of code in Makefile.aldec

binupr, 2020-02-25:
Hi,

While debugging a libcocotbvhpi.dll missing issue for cocotb1.3, I noticed that some parts of the Makefile.aldec is incomplete. Like the below:
--
else ifeq ($(TOPLEVEL_LANG),vhdl)
--
is anyone working on this?

We recently migrated to cocotbv1.3 and unable to get our Aldec RivieraPro simulations progressing due to the libcocotbvhpi.dll missing issue. In that debugging process, I noticed the Makefile.aldec has some gotchas which might be root cause.

Cheers
Binu

cmarqu, 2020-02-25:
(Re the ``libcocotbvhdl.dll`` problem, see https://github.com/cocotb/cocotb/issues/1459#issuecomment-590612165)

cmarqu, 2020-02-25:
Not sure what you mean with "incomplete".
There is an 
* "if" branch for ``TOPLEVEL_LANG=verilog`` (https://github.com/cocotb/cocotb/blob/v1.3.0/cocotb/share/makefiles/simulators/Makefile.aldec#L84),
* an "elseif" for ``TOPLEVEL_LANG=vhdl`` (https://github.com/cocotb/cocotb/blob/v1.3.0/cocotb/share/makefiles/simulators/Makefile.aldec#L90)
* and an "else" for reporting an error when ``TOPLEVEL_LANG`` is neither of those (https://github.com/cocotb/cocotb/blob/v1.3.0/cocotb/share/makefiles/simulators/Makefile.aldec#L103).

cmarqu, 2020-02-25:
By the way, @binupr and @mao-karin, after this problem is resolved (hopefully soon), could you help us verifying that #1378 works as intended?

mao-karin, 2020-02-26:
Hi yes I can look into this afterwards.

binupr, 2020-02-26:
> Not sure what you mean with "incomplete".
> There is an
>
> * "if" branch for `TOPLEVEL_LANG=verilog` (https://github.com/cocotb/cocotb/blob/v1.3.0/cocotb/share/makefiles/simulators/Makefile.aldec#L84), [...]

I was getting confused with the branching of that "else ifeq". Agree that Line90 was complete. But I would propose the below change with appropriate tabbing for clarity. I test this locally and works on Riviera Pro:
=====
else ifeq ($(TOPLEVEL_LANG),vhdl)

	ifeq ($(OS),Msys)
	    VHPI_LIB = $(shell sh -c 'cd $(LIB_DIR) && pwd -W')/libcocotbvhpi
	else
	    VHPI_LIB = $(LIB_DIR)/libcocotbvhpi
	endif
	    GPI_ARGS = -loadvhpi $(VHPI_LIB):vhpi_startup_routines_bootstrap
	ifneq ($(VERILOG_SOURCES),)
	    GPI_EXTRA = vpi
	endif

else
   $(error "A valid value (verilog or vhdl) was not provided for TOPLEVEL_LANG=$(TOPLEVEL_LANG)")
endif
===

cmarqu, 2020-03-08:
@binupr I have edited your comment above and wrapped the Makefile code example with triple backticks - is that what you were proposing?
In general, we have had bugs due to innocent looking Makefile formatting (tabs vs spaces; extra space; comments at the end of a line) in the past (some only showing up with certain code paths), so I'm wary of doing that edit even if it looks fine...

eric-wieser, 2020-04-09:
So to be clear, the request is just to make the indentation clearer here:

https://github.com/cocotb/cocotb/blob/2d11eeac80e97f2f356662b029228690dfc990d5/cocotb/share/makefiles/simulators/Makefile.aldec#L84-L105

Note that it's very important to use spaces and not tabs for this type of indent, else the makefile will do the wrong thing.