cocotb/cocotb #1461: A tcl based flow ?

themperek, 2020-02-26:
We could now make a simple tcl based flow. 

A Modelsim script for adder example (`vsim -c -do run.do`):
```tcl
# run.do

# General cocotb settings
exec cocotb-config --prefix > /tmp/cocotb-prefix
set COCOTB_DIR [string trim [read [open "/tmp/cocotb-prefix" r]]]
set VPI_LIB $COCOTB_DIR/cocotb/libs/modelsim/libvpi.so
set ::env(COCOTB_SIM) 1
set ::env(PYTHONPATH) ${COCOTB_DIR}/cocotb/libs/modelsim

# This test settings
set ::env(MODULE) test_adder
set ::env(PYTHONPATH) ./:../model:$::env(PYTHONPATH)

# Compile and run
vlog ../hdl/adder.v
vsim  -pli $VPI_LIB adder
run -all
```

Should we add such flow under examples (can use Icarus in CI) or document it?

garmin-mjames, 2020-02-26:
I ended up making a tcl package with some helper procedures that set the `::env` vars and launch simulation with the right library.
Using `TCLLIBPATH` envar to point to the package folder, your script can then use `package require cocotb` and call the procedures. Keeps the script clean, and less brittle if the underlying details change.

garmin-mjames, 2020-03-26:
You can get the output of exec directly, btw.
Also using **file** commands is platform agnostic:
```tcl
if {[catch {set COCOTB_DIR [file normalize [exec cocotb-config --tcl]]}]} {
    puts "cocotb-config failed to execute"
    puts "  $::errorInfo"
    exit -code 1
    }
set VPI_LIB [file join $COCOTB_DIR cocotb libs modelsim libcocotbvpi.so]
...
```

themperek, 2020-04-27:
@garmin-mjames thank you for the hint.

I close in favor of https://github.com/cocotb/cocotb/issues/1718

closed 2020-04-27 by themperek