cocotb/cocotb #1463: Python 3.8.1 installation error

tsengr0916, 2020-02-27:
I'm trying to use Python 3.8.1 to pip install COCOTB on Linux Suse12 machine. I got the error like this, "ModuleNotFoundError: No module named '_ctypes'". 

Any ideas? 

The trace of the error is like below:

Processing /prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/coco3/cocotb
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
ERROR: Exception:
Traceback (most recent call last):
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/commands/install.py", line 331, in run
    resolver.resolve(requirement_set)
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 515, in prepare_linked_requirement
    abstract_dist = _get_prepared_distribution(
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/operations/prepare.py", line 95, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 151, in get_requires_for_build_wheel
    return self._call_hook('get_requires_for_build_wheel', {
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/coco3/venv3.8.1/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 63, in _build_backend
    obj = import_module(mod_path)
  File "/pkg/qct/software/python/sles12/3.8.1/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/tmp/pip-build-env-loac6991/overlay/lib/python3.8/site-packages/setuptools/__init__.py", line 19, in <module>
    from setuptools.dist import Distribution, Feature
  File "/tmp/pip-build-env-loac6991/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 36, in <module>
    from setuptools import windows_support
  File "/tmp/pip-build-env-loac6991/overlay/lib/python3.8/site-packages/setuptools/windows_support.py", line 2, in <module>
    import ctypes
  File "/pkg/qct/software/python/sles12/3.8.1/lib/python3.8/ctypes/__init__.py", line 7, in <module>
    from _ctypes import Union, Structure, Array
ModuleNotFoundError: No module named '_ctypes'
 
I don't have this issue with Python 3.5. 
Your help/input will be highly appreciated!

Thanks,
Richard

eric-wieser, 2020-02-28:
You seen to have a broken python installation, as your error is coming from trying to import the builtin python `ctypes` module.

tsengr0916, 2020-02-28:
From this link https://bugs.python.org/issue31652, it seems we need to install  libffi-devel. Is that really true? 
I checked our Python 3.5 installation, which works with COCOTB. It has the similar code to import _ctypes at the op of the ctypes.

themperek, 2020-02-28:
This is streange. We have 3.8 in CI. What you can try is to add `--no-use-pep517` during installation see http://pip.pypa.io/en/stable/reference/pip/

eric-wieser, 2020-02-28:
It might be worth opening the same issue against pypa/setuptools, and see if they have any thoughts.

Millennium-Software-Design, 2020-02-28:
I ran into this error when I built python 3.8 myself.  You will need libffi installed before you build python (assuming that's what's going on here).

themperek, 2020-03-03:
@tsengr0916 Could you provide any update on this issue or close it?

themperek, 2020-03-22:
@tsengr0916 I close this since no update provided. Please reopen if you still have an issue.

closed 2020-03-22 by themperek