cocotb/cocotb #1466: Documentation building: "WARNING: duplicate label"

cmarqu, 2020-02-29:
With ``sphinx.ext.autosectionlabel`` (in use since https://github.com/cocotb/cocotb/commit/862012b3c01f90ea1fb715c206b4eb785119f964), we got quite some new "duplicate label" warnings.

To fix some of those warnings, we could set ``autosectionlabel_prefix_document = True`` which prefixes each automatically generated section label with the name of the document it is in, followed by a colon.

A drawback is that we would now hardcode the file name of the document into the reference,
so that we are not as free with renaming documents anymore. Also, we cannot catch all those duplicates that way, e.g. in the release notes, there are multiple "new features" subsections.

The alternative is to remove ``sphinx.ext.autosectionlabel`` and set our own reference anchors.
I would prefer this.

Ping @Martoni

Martoni, 2020-03-01:
Hi Colin,

> 
> The alternative is to remove ``sphinx.ext.autosectionlabel`` and set our own reference anchors.
> I would prefer this.

If I remember, I added this dependency just to be able to reference
«chapter». Then yes I think we can remove the dependency and add an
anchors (i'm a sphinx beginner ;).

Do you want me to do it ?

Regards,
Martoni
http//www.fabienm.eu/flf/

cmarqu, 2020-03-01:
Hi Fabien,

Thanks for the quick reply, I can do it as part of some other changes, I just wanted to get your opinion on it.

PR #1470 (2020-03-02): Remove Sphinx extension autosectionlabel
Introduce manual reference anchors and use them.

Closes #1466.
closed 2020-03-02 by cmarqu