cocotb/cocotb #1476: test_force_release casuing regression on Questa with FLI

ktbarrett, 2020-03-03:
Introduced #1403.

```
 $ make SIM=questa COMPILE_ARGS="+acc" SIM_ARGS="-t ps" TOPLEVEL_LANG=vhdl
make results.xml
make[1]: Entering directory `.../cocotb/tests/test_cases/test_force_release'
mkdir -p sim_build
set -o pipefail; LD_LIBRARY_PATH=.../lib/python3.7/site-packages/cocotb/libs/modelsim: MODULE=test_force_release TESTCASE= TOPLEVEL="work.sample_module" COCOTB_SIM=1 \
GPI_EXTRA= TOPLEVEL_LANG=vhdl PYTHONPATH=.../lib/python3.7/site-packages/cocotb/libs/modelsim:.../cocotb/tests/test_cases/test_force_release:.../lib/python3.7/site-packages:... \
.../bin/vsim -c -64  -do sim_build/runsim.do  2>&1 | tee sim_build/sim.log
Reading pref.tcl

# 2019.1

# do sim_build/runsim.do
# QuestaSim-64 vmap 2019.1 Lib Mapping Utility 2019.01 Jan  1 2019
# vmap -c 
# Copying .../linux_x86_64/../modelsim.ini to modelsim.ini
# QuestaSim-64 vmap 2019.1 Lib Mapping Utility 2019.01 Jan  1 2019
# vmap work sim_build/work 
# Modifying modelsim.ini
# QuestaSim-64 vcom 2019.1 Compiler 2019.01 Jan  1 2019
# Start time: 16:46:50 on Mar 03,2020
# vcom -work work "+acc" .../cocotb/tests/test_cases/test_force_release/../../../tests/designs/sample_module/sample_module_pack.vhdl .../cocotb/tests/test_cases/test_force_release/../../../tests/designs/sample_module/sample_module_1.vhdl .../cocotb/tests/test_cases/test_force_release/../../../tests/designs/sample_module/sample_module.vhdl 
# -- Loading package STANDARD
# -- Loading package TEXTIO
# -- Loading package std_logic_1164
# -- Loading package NUMERIC_STD
# -- Compiling package sample_module_pack
# -- Compiling package body sample_module_pack
# -- Loading package sample_module_pack
# -- Compiling entity sample_module_1
# -- Compiling architecture impl of sample_module_1
# -- Loading package sample_module_pack
# -- Compiling entity sample_module
# -- Compiling architecture impl of sample_module
# End time: 16:46:50 on Mar 03,2020, Elapsed time: 0:00:00
# Errors: 0, Warnings: 0
# vsim -t ps -onfinish exit -foreign "cocotb_init .../lib/python3.7/site-packages/cocotb/libs/modelsim/libfli.so" sim_build/work.sample_module 
# Start time: 16:46:50 on Mar 03,2020
# ** Note: (vsim-3812) Design is being optimized...
# //  Questa Sim-64
# //  Version 2019.1 linux_x86_64 Jan  1 2019
# //
# //  Copyright 1991-2019 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading std.standard
# Loading std.textio(body)
# Loading ieee.std_logic_1164(body)
# Loading ieee.numeric_std(body)
# Loading work.sample_module_pack(body)
# Loading work.sample_module(impl)#1
# Loading work.sample_module_1(impl)#1
# Loading .../lib/python3.7/site-packages/cocotb/libs/modelsim/libfli.so
#      -.--ns INFO     cocotb.gpi                         ..ib/fli/FliImpl.cpp:1077 in cocotb_init                     cocotb_init called
# 
#      -.--ns INFO     cocotb.gpi                         ../embed/gpi_embed.c:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
#      -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:104  in gpi_print_registered_impl       FLI registered
#      0.00ns INFO     cocotb                                      __init__.py:144  in _initialise_testbench           Running on ModelSim for Questa-64 version 2019.1 2019.01
#      0.00ns INFO     cocotb                                      __init__.py:145  in _initialise_testbench           Python interpreter initialized and cocotb loaded!
#      0.00ns INFO     cocotb                                      __init__.py:163  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from .../lib/python3.7/site-packages
#      0.00ns INFO     cocotb                                      __init__.py:180  in _initialise_testbench           Seeding Python random module with 1583275611
#      0.00ns INFO     cocotb.regression                         regression.py:189  in initialise                      Found test test_force_release.test_force_release
#      0.00ns INFO     cocotb.regression                         regression.py:417  in execute                         Running test 1/1: test_force_release
#      0.00ns INFO     ..test_force_release.0x7ffff1543ef0       decorators.py:253  in _advance                        Starting test: "test_force_release"
#                                                                                                                      Description: 
#                                                                                                                          Test force and release on simulation handles
#                                                                                                                          
FATAL: We are calling up again
U/I: Unexpected EOF on RPC channel sock2dc2490

** Fatal: Trouble with Simulation Kernel.
#     10.00ns CRITICAL cocotb.gpi                                FliObjHdl.cpp:328  in set_signal_value                Force or release action not supported for FLI.
make[1]: *** [results.xml] Error 218
make[1]: Leaving directory `.../cocotb/tests/test_cases/test_force_release'
make: *** [sim] Error 2
```

thasti, 2020-03-04:
Do you see the regression here in that `Force or release action not supported for FLI.` or in that the scheduler bugs out (`FATAL: We are calling up again`)?

If it is the latter: I indeed saw the same message yesterday also in Incisive (using VPI, against an actual simulation using Force/Release, not this testcase) but didn't bisect yet whether #1403 is the actual root cause of it.

ktbarrett, 2020-03-04:
@thasti "FATAL: We are calling up again" I see as the error. #1403 is when it started happening, I don't the cause yet.

thasti, 2020-03-09:
I managed to bisect this. For me, it is 628dfa8038178236c1bad8631503f7b53b664e53 that introduces the breakage. Every simulation in Incisive (even without Force/Release) fails with 

```
FATAL: We are calling up again
```

c4524ebfa73e72c3452084228ffef3337adce353 still works fine, so #1403 is not the culprit here. 

Even the most simple test case triggers this for me:
```
import cocotb
from cocotb.triggers import Timer

@cocotb.test()
def test_case(dut):
    yield Timer(1, "ns")
```

@ktbarrett could you confirm that the same is true for Questa?

@themperek Any ideas? I cross-checked that this does not happen with iverilog, but have no other simulators nearby at the moment.

thasti, 2020-03-09:
Another piece of (perhaps related) evidence:

Starting with 628dfa8, on startup I see the following. It appears VPI is registered twice? 
```
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     -.--ns WARNING  cocotb.gpi                         ../gpi/GpiCommon.cpp:117  in gpi_register_impl               VPI already registered, check GPI_EXTRA
ERROR:          VPI       TFTHRE
         The user defined task or function $info has already been registered.
ERROR:          VPI       TFTHRE
         The user defined task or function $warning has already been registered.
ERROR:          VPI       TFTHRE
         The user defined task or function $error has already been registered.
ERROR:          VPI       TFTHRE
         The user defined task or function $fatal has already been registered.
```

ktbarrett, 2020-03-09:
@thasti 
This happens for Questa when running with the toplevel language as VHDL, so it can't be the rename of libvpi. And it only happens for this test. I'll look at it soon.

ktbarrett, 2020-04-21:
Apparently this isn't an issue? I recently run from #1656 shows this test passing.

ktbarrett, 2020-05-16:
It works fine in 10.7c, but testing with 2020.1 results in a crash and a hang, and 2019.3 shows a crash.

Here is the backtrace:
```
#0  0x00007ffff251aff6 in gpi_get_sim_time (high=0x7fffffff6410, low=0x7fffffff6414) at cocotb/share/lib/gpi/GpiCommon.cpp:231
#1  0x00007ffff7f03895 in get_sim_time (self=<optimized out>, args=<optimized out>) at cocotb/share/lib/simulator/simulatormodule.cpp:731
#2  0x00007ffff216215a in _PyMethodDef_RawFastCallKeywords () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#3  0x00007ffff2162ba6 in _PyCFunction_FastCallKeywords () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#4  0x00007ffff20b43df in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#5  0x00007ffff20b210a in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#6  0x00007ffff20b2e30 in function_code_fastcall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#7  0x00007ffff20b40f6 in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#8  0x00007ffff20af808 in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#9  0x00007ffff20b2e30 in function_code_fastcall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#10 0x00007ffff20b40f6 in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#11 0x00007ffff20adcb9 in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#12 0x00007ffff20b2e30 in function_code_fastcall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#13 0x00007ffff20b40f6 in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#14 0x00007ffff20adcb9 in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#15 0x00007ffff20b2e30 in function_code_fastcall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#16 0x00007ffff20b40f6 in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#17 0x00007ffff20adcb9 in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#18 0x00007ffff20b2e30 in function_code_fastcall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#19 0x00007ffff20b40f6 in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#20 0x00007ffff20adcb9 in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#21 0x00007ffff20b2e30 in function_code_fastcall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#22 0x00007ffff20b40f6 in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#23 0x00007ffff20adcb9 in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#24 0x00007ffff20b2e30 in function_code_fastcall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#25 0x00007ffff2162a2f in _PyFunction_FastCallDict () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#26 0x00007ffff21634c3 in object_vacall () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#27 0x00007ffff2163689 in PyObject_CallFunctionObjArgs () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#28 0x00007ffff251294f in gpi_log_v (argp=0x7fffffff7388, msg=0x7ffff23ae150 "FATAL: We are calling up again", lineno=340, funcname=0x7ffff23ae570 <embed_sim_event::__func__> "embed_sim_event", 
    pathname=0x7ffff23ae128 "cocotb/share/lib/embed/gpi_embed.cpp", level=GPIError, name=0x7ffff23ae011 "cocotb.gpi") at cocotb/share/lib/gpi_log/gpi_logging.cpp:237
#29 gpi_log (name=0x7ffff23ae011 "cocotb.gpi", level=GPIError, pathname=0x7ffff23ae128 "cocotb/share/lib/embed/gpi_embed.cpp", funcname=0x7ffff23ae570 <embed_sim_event::__func__> "embed_sim_event", lineno=340, 
    msg=0x7ffff23ae150 "FATAL: We are calling up again") at cocotb/share/lib/gpi_log/gpi_logging.cpp:264
#30 0x00007ffff23adf36 in embed_sim_event (level=<optimized out>, msg=<optimized out>) at cocotb/share/lib/embed/gpi_embed.cpp:340
#31 0x00007ffff251b715 in gpi_embed_end () at cocotb/share/lib/gpi/GpiCommon.cpp:134
#32 0x00007ffff25384d9 in FliShutdownCbHdl::run_callback (this=<optimized out>) at cocotb/share/lib/fli/FliCbHdl.cpp:219
#33 0x00007ffff2533331 in handle_fli_callback (data=0x37e95a0) at cocotb/share/lib/fli/FliImpl.cpp:1027
#34 0x00000000009dd66f in ?? ()
#35 0x0000000000dfa684 in ?? ()
#36 0x00000000009fb7b5 in ?? ()
#37 0x00007ffff6f17cd9 in __run_exit_handlers () from /lib64/libc.so.6
#38 0x00007ffff6f17d27 in exit () from /lib64/libc.so.6
#39 0x00007ffff253aaf6 in FliLogicObjHdl::set_signal_value (this=<optimized out>, value=<optimized out>, action=<optimized out>) at cocotb/share/lib/fli/FliObjHdl.cpp:321
#40 0x00007ffff7f03c48 in set_signal_val_long (self=0x7ffff1e07b58, args=<optimized out>) at cocotb/share/lib/simulator/simulatormodule.cpp:629
#41 0x00007ffff2162c83 in PyCFunction_Call () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
#42 0x00007ffff20b1d61 in _PyEval_EvalFrameDefault () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
[... cut ...]
#56 0x00007ffff20b40f6 in call_function.lto_priv.1536 () from .../anaconda/anaconda/2019.03/install/lib/libpython3.7m.so.1.0
```

As you can see, `LOG_CRITICAL` is used to inform the user that `Force` and `Release` aren't supported on Questa. `LOG_CRITICAL` calls `exit()`, perhaps this was intended. `exit` calls the shutdown callback. In effect making the GPI re-entrant. This is why "we are calling up again" is printed and eventually leads to a returning down the stack into Python post-finalization causing the segfault.

We probably shouldn't be using `LOG_CRITICAL` there.

PR #1863 (2020-06-03): cleanup usage of LOG_CRITICAL in GPI and implementations
xref #1844. Should fix #1476.

`LOG_CRITICAL` calls `exit`. On Questa 2019 and 2020, `exit` calls the registered shutdown callback immediately. This causes GPI re-entrance from the simulator. The re-entrant call shuts down cocotb and finalizes the interpreter, but the original caller of `exit` might return through now invalid Python code. `LOG_CRITICAL` is *never* a good idea in user-called code. To remedy this, all calls to `LOG_CRITICAL` in user-called code were replaced with `LOG_ERROR` and a error code returned.

For simulator-called [... cut]
closed 2020-06-03 by eric-wieser