cocotb/cocotb #1477: Partition environment variables

ktbarrett, 2020-03-04:
From https://github.com/cocotb/cocotb/pull/1293#discussion_r386031732.

In the [Build options and Environment Variables]( https://cocotb.readthedocs.io/en/latest/building.html#build-options-and-environment-variables) section of the the documentation, there are a lot of environment variables under the sections: "Make Variables", "Environment Variables", and "More Environment Variables", which aren't terribly useful names or ways of partitioning the variables.

They should be broken into several partitions according to the consumer/owner of the variable: GPI, Python/GPI interface, the regression manager, the scheduler, and the makefile-based test scripts, etc.

cmarqu, 2020-03-07:
I agree. In order to ease collaboration on the partitioning, I have taken the names and short description from ``Makefile.sim`` and copied them to https://github.com/cocotb/cocotb/wiki/__Playground__

cmarqu, 2020-03-08:
Of the list, I would pull the variables marked in <https://github.com/cocotb/cocotb/wiki/__Playground__> with ``(__common__)`` into a beginners tutorial in #1482 (or follow-up PR).

ktbarrett, 2020-03-10:
Updated the wiki, added `RANDOM_SEED`.

PR #1729 (2020-05-14): Partition make- and environment variables
Closes #1477
commit 53740b79 (2020-05-14): Partition make- and environment variables

Closes #1477
commit dd30e230 (2020-05-14): Partition make- and environment variables

Closes #1477
closed 2020-05-14 by cmarqu