cocotb/cocotb #1478: cocotb verilator code coverage support

lavanyajagan, 2020-03-04:
Could you add verilator coverage support in the following file:

``cocotb/share/lib/verilator/verilator.cpp``

```
diff --git a/cocotb/share/lib/verilator/verilator.cpp b/cocotb/share/lib/verilator/verilator.cpp
index aff06d9..fc43374 100644
--- a/cocotb/share/lib/verilator/verilator.cpp
+++ b/cocotb/share/lib/verilator/verilator.cpp
@@ -84,5 +84,9 @@ int main(int argc, char** argv) {
     tfp->close();
 #endif
 
+#if VM_COVERAGE
+    VerilatedCov::write("coverage.dat");
+#endif
+
     return 0;
 }
```

imphil, 2020-03-04:
Thank you for your issue. Would you mind opening a pull request with the proposed change?

themperek, 2020-03-04:
We already have `COVERAGE` in Aldec but the same variable is used to collect python coverage so maybe we should use different variables.

imphil, 2020-03-04:
VM_COVERAGE is a Verilator define which is set if Verilator is instructed to collect coverage (when compiling the simulation).

This is different from a variable that users would use in a cocotb makefile to instruct Verilator to actually build in coverage (the equivalent to `COVERAGE` in Aldec)

lavanyajagan, 2020-03-04:
https://github.com/cocotb/cocotb/pull/1479 

Hope this is sufficient. Thanks

closed 2020-03-26 by cmarqu

sammy17, 2021-08-22:
I noticed that even when I enable verilator code coverage with `EXTRA_ARGS= --coverage`, coverage.dat file is not getting generated even with the above fix (as mentioned in https://github.com/cocotb/cocotb/issues/2573#issuecomment-865454544). 

After a quick debug I noticed that anything after `vl_finish(__FILE__, __LINE__, "");` in `cocotb/share/lib/verilator/verilator.cpp` is not getting executed. So `VerilatedCov::write("coverage.dat");` that was added for this issue never got executed and `coverage.dat` file was not dumped.

I was able to get the coverage file to be dumped by moving the `VerilatedCov::write` call right before `vl_finish` as below.

```
        // If there are no more cbAfterDelay callbacks,
        // the next deadline is max value, so end the simulation now
        if (next_time == static_cast<vluint64_t>(~0ULL)) {
#if VM_COVERAGE
            VerilatedCov::write("coverage.dat");
#endif
            vl_finish(__FILE__, __LINE__, "");
            break;
        } else {
            main_time = next_time;
        }
```

forensicgarlic, 2023-01-27:
I just noticed the same issue as @sammy17 above on Cocotb 1.7.1 with Verilator 5, and applying the same fix worked for me.