cocotb/cocotb #1480: An unexpected termination has occurred in sim_build/simv due to a signal: Segmentation fault

futurehome, 2020-03-05:
Hi,

I tried to run an example test - axi_lite_slave, with cocotb 1.3, vcsmx/2018.09_sp2_8, but the simulation crashed, I found something unusual in the log file:
1)     -.--ns ERROR    cocotb.gpi                                 VpiCbHdl.cpp:74   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbStartOfSimulation(11)
2)     -.--ns ERROR    cocotb.gpi                                 VpiCbHdl.cpp:74   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbEndOfSimulation(12)
3) The VPI function 'vpi_chk_error' is not supported in the compiler.
4) An unexpected termination has occurred in sim_build/simv due to a signal: Segmentation fault.

Please find the detail in the attached log file.
Can you please help to figure it out?

Thanks,
Alex

[axi_lite_slave.log](https://github.com/cocotb/cocotb/files/4290643/axi_lite_slave.log)

cmarqu, 2020-03-05:
Related: #1261

futurehome, 2020-03-06:
Are there any ways that I can use to avoid this bug temporarily?

Alex

cmarqu, 2020-03-06:
Try the change described in https://github.com/cocotb/cocotb/issues/1261#issuecomment-571718339 which is here: https://github.com/cocotb/cocotb/blob/77ecf0ef67aad31916be1baf16b8174e8591b70d/cocotb/share/lib/gpi/GpiCommon.cpp#L137

futurehome, 2020-03-09:
Still the same issue after modify GpiCommon.cpp file.
[image]

cmarqu, 2020-03-09:
One pitfall that may be the cause: if you installed cocotb with pip (in editable mode), you need to re-run pip with the ``-f`` option in order to actually compile the C++ code change.

themperek, 2020-03-10:
>One pitfall that may be the cause: if you installed cocotb with pip (in editable mode), you need to re-run pip with the -f option in order to actually compile the C++ code change.

@cmarqu This is the only case for >= 1.4

closed 2021-03-14 by github-actions[bot]