cocotb/cocotb #1487: Resolve issue with invalid COCOTB_RESOLVE_X value

garmin-mjames, 2020-03-10:
https://github.com/cocotb/cocotb/blob/e9490fab386476fa977f7c7da28baec2555530ce/cocotb/binary.py#L36-L53

Currently, when `COCOTB_RESOLVE_X` is not set to one of the supported options, the `resolve` function does nothing and does not raise an error, returning the unresolved binary string.

I think we should default to `VALUE_ERROR` behavior rather than blowing up the integer conversion, and possibly log an info/warning message when we check `COCOTB_RESOLVE_X`.

garmin-mjames, 2020-03-16:
Separately but related, `RANDOM` resolves all instances of each value to the same 0 or 1.
I.e.,
all `X`'s become `0`, all `x`'s become '1', etc.

It would be better to have each bit be resolved separately, so `XXXX` wouldn't be forced to `0000` or `1111`.

ktbarrett, 2020-07-16:
One more issue is that `-` (AKA "don't cares") are resolved to 0 instead of being treated the same as `X`.

ktbarrett, 2025-02-17:
All of these issues and the incorrect translation of VHDL's `L` and `H` values have been fixed on master across several PRs.