cocotb/cocotb #1490: Decide on stable branch names

imphil, 2020-03-11:
We need a branch for stable/maintenance releases, e.g. 1.3.x.

My recommendations:
- The branch name should not overlay the version tag namespace, which is currently starting with `v`, e.g. `v1.3.0`. This makes it easier to auto-complete, and easier to set up GitHub branch filters.
- I'd like to avoid the full word `maintenance` as I'm always misspelling it.

Suggestions:
- stable-1.3
- stable/1.3
- maint-1.3
- stable/1.3

eric-wieser, 2020-03-11:
Argument for slashes - my graphical git client groups together branches named in that way. It also maps to folders within `git/refs/heads`

themperek, 2020-03-11:
The decision is: `stable/1.3` the branch is here: https://github.com/cocotb/cocotb/tree/stable/1.3

closed 2020-03-11 by themperek