cocotb/cocotb #1494: VHPI callback issue in Active-HDL 10.4 through 10.5a

garmin-mjames, 2020-03-11:
There is a bug in several versions of Aldec Active-HDL that prevent cocotb from working.

Simplified, the issue is as follows:
The VHPI callback used for cocotb's `ReadWrite` trigger is not called on delta cycles of a time step.

So this test will not work:
```
@cocotb.test()
def test_signal_set(dut):
    cocotb.fork(Clock(dut.clk, 10, units='ns').start())
    yield RisingEdge(dut.clk)
    dut.sig1 <= 1
```
The first `ReadWrite` callback that sets the clock value will work. The `RisingEdge` callback also works, and is triggered on the next delta cycle. However, the `ReadWrite` callback where `dut.sig1` will be set is never called.

### Versions affected: ###
- Aldec Active-HDL 10.4
- Aldec Active-HDL 10.4a
- Aldec Active-HDL 10.5
- Aldec Active-HDL 10.5a

I submitted a support ticket, and it was fixed in Active-HDL 11.1.
From 11.1 Release Notes:
```
Problems Corrected in Version 11.1
An issue with the vhpiCbRepEndOfProcesses callback from the VHPI interface was resolved. (SPT78258)
```

### Followup ###
- [ ] Add Active-HDL to [Simulator Support](https://cocotb.readthedocs.io/en/latest/simulator_support.html) documentation and note this limitation.

cmarqu, 2020-04-09:
We could even do a check for these versions in cocotb - IIRC @ktbarrett wanted to expose getting simulator name and version from Python.

garmin-mjames, 2020-04-09:
The check is easy in Tcl:
```tcl
proc getActiveVersion {} {
    set versionList [split $version { .}]
    if {[lindex $versionList 0] == "build"} {
        set versionList [lrange $versionList 1 end]
    }
    return [join [lrange $versionList 0 1] "."]
}
```
Supports both Active-HDL GUI and VSimSA. VSimSA puts "build" at the beginning:
VSimSA:
```
$puts $version
build 10.1.3088.5434
```
Active-HDL GUI:
```
# puts $version
# 10.1.3088.5434
```

ktbarrett, 2020-05-27:
We already had the ability to check the simulator product and version in cocotb. They are available in cocotb via `cocotb.SIM_NAME` and `cocotb.SIM_VERSION`.

imphil, 2023-05-19:
This issue is resolved in newer versions of ActiveHDL and mentioned in the simulator support section of the cocotb documentation. I'm therefore closing this issue, there's nothing to be done on our side.

closed 2023-05-19 by imphil