cocotb/cocotb #1495: Deprecate simulator-specific make variables

cmarqu, 2020-03-11:
The maintainers decided to deprecate some make variables that have not been implemented for all simulators. Reasons are the general maintenance burden, and that we often cannot test the effect of these variables in CI.
If any of these variables are used, we will print a warning with a reference to the documentation where we will explain how to achieve the same effect in a different way.

The initial list:
- [ ] Deprecate ``LICENSE_QUEUE``
- [ ] Deprecate ``COCOTB_NVC_TRACE``
- [ ] Deprecate ``VERILATOR_SIM_DEBUG``
- [ ] Deprecate ``VERILATOR_TRACE``

To be decided:
- [ ] Deprecate ``WAVES``
- [ ] Deprecate ``SCRIPT_FILE``

themperek, 2020-04-09:
Also all `V*_ARG` and `A*_LOG`?

alexforencich, 2020-11-29:
We should keep WAVES, but consider renaming it and implement it for all simulators.  Same with basic support for module parameters, perhaps at the top-level only.  Anything more simulator-specific than that can then be done manually, if necessary.

imphil, 2024-07-08:
Removing 1.9 milestone: We're done with changes for 1.9.0, anything else needs to go in a future release.